tgalarneau/bms #3: Setup clarification - Uncomment sock commands

the-butterfry, 2021-04-28:
First time with all the Tele/Influx/Graf stack so bear with me..

I believe everthing is flowing correctly. I can see the grafna dashboard make requests in the syslog. I got the script working manually, and i can see the cell data scroll through (i only have 8 cells, I'll deal with that later) 

`pi@Farm-pi:~/bms $ sudo /usr/local/opt/python-3.9.4/bin/python3.9 jbdbms.py -b A4:C1:38:EC:78:4F -i 5 -m jbdbms
connect 1
sending
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
jbdbms,56580,16,3356,3357,3360,3358,3358,3357
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
jbdbms,56580,16,3356,3355,3360,3358,3359,3356`

but the problem is I'm not seeing any data coming through.

You mention briefly in the readme about uncommenting the sock commands but not down in the configuration section. 

here's what i did, do i need to do something specific to my setup?

` telegraf_socket = "/tmp/telegraf.sock"                                          # The file handler for the Telegraf process.
sock = socket.socket(socket.AF_UNIX, socket.SOCK_DGRAM)         # Connection to Telegraf, over a network socket.
sock.connect(telegraf_socket) `
 
do i need to uncomment all the #sock.send(message.encode('utf8')) lines too?

tgalarneau, 2021-04-28:
What I have is :

telegraf_socket = "/tmp/telegraf.sock"
sock = socket.socket(socket.AF_UNIX, socket.SOCK_DGRAM)
sock.connect(telegraf_socket)

Look at systemctl status telegraf
systemctl status influxdb

You can run telegraf with -debug for more info

Check Grafana that the data queries point to your database.

tgalarneau, 2021-04-28:
also looks like your data is out of order. The cvs headers should be
followed by correct data. Your data seems to mix cell data with info data.
Maybe because you have not corrected format for 8 cells instead of 16.

the-butterfry, 2021-04-28:
thanks.. so yeah everyting looks ok there. 

influx status:
pi@Farm-pi:~/bms $ sudo service influxdb status
● influxdb.service - InfluxDB is an open-source, distributed, time series database
   Loaded: loaded (/lib/systemd/system/influxdb.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-04-28 16:01:12 PDT; 48min ago
     Docs: https://docs.influxdata.com/influxdb/
 Main PID: 16500 (influxd)
    Tasks: 14 (limit: 3858)
   CGroup: /system.slice/influxdb.service
           └─16500 /usr/bin/influxd -config /etc/influxdb/influxdb.conf

Apr 28 16:48:15 Farm-pi influxd[16500]: [httpd] ::1 - solar [28/Apr/2021:16:48:15 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 3381bea7-a87c-11eb-88b7-e45f01107201 22425
Apr 28 16:48:25 Farm-pi influxd[16500]: [httpd] ::1 - solar [28/Apr/2021:16:48:25 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 3977bedd-a87c-11eb-88b8-e45f01107201 15930
Apr 28 16:48:31 Farm-pi influxd[16500]: [httpd] ::1 - solar [28/Apr/2021:16:48:31 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 3ca3dd7e-a87c-11eb-88b9-e45f01107201 15362
Apr 28 16:48:54 Farm-pi influxd[16500]: ts=2021-04-28T23:48:54.460235Z lvl=info msg="Executing query" log_id=0TngHd0l000 service=query query="CREATE DATABASE battery"
Apr 28 16:48:54 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:48:54 -0700] "POST /query HTTP/1.1 {'q': 'CREATE DATABASE "battery"'}" 200 57 "-" "Telegraf/1.18.1 Go/1.16.2" 4a856358-a87c-11eb-88ba-e45f01107201 1427
Apr 28 16:49:04 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:49:04 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 507bed46-a87c-11eb-88bb-e45f01107201 20605
Apr 28 16:49:14 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:49:14 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 5671db98-a87c-11eb-88bc-e45f01107201 8240
Apr 28 16:49:24 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:49:24 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 5c67e4c3-a87c-11eb-88bd-e45f01107201 15587
Apr 28 16:49:34 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:49:34 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 625db93f-a87c-11eb-88be-e45f01107201 9736
Apr 28 16:49:44 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:49:44 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 6853e019-a87c-11eb-88bf-e45f01107201 13029

syslog:
pi@Farm-pi:~/bms $ sudo tail -f /var/log/syslog
Apr 28 16:50:48 Farm-pi sudo[17486]: jbdbms,56580,16,3353,3351,3353,3355,3356,3353
Apr 28 16:50:48 Farm-pi sudo[17486]: sending
Apr 28 16:50:48 Farm-pi sudo[17486]: meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
Apr 28 16:50:48 Farm-pi sudo[17486]: jbdbms,56580,16,3353,3354,3356,3355,3356,3354
Apr 28 16:50:48 Farm-pi sudo[17486]: sending
Apr 28 16:50:48 Farm-pi sudo[17486]: meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
Apr 28 16:50:48 Farm-pi sudo[17486]: jbdbms,56580,16,3354,3352,3354,3355,3357,3353
Apr 28 16:50:48 Farm-pi sudo[17486]: sending
Apr 28 16:50:48 Farm-pi sudo[17486]: meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
Apr 28 16:50:54 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:50:54 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 920da4e6-a87c-11eb-88c6-e45f01107201 27673
Apr 28 16:51:04 Farm-pi influxd[16500]: [httpd] 127.0.0.1 - solar [28/Apr/2021:16:51:04 -0700] "POST /write?db=battery HTTP/1.1 " 204 0 "-" "Telegraf/1.18.1 Go/1.16.2" 9803b0c0-a87c-11eb-88c7-e45f01107201 16300

as for the -m option.. i don't have a monitor. So.. is that an issue?

the-butterfry, 2021-04-29:
ahh, ok getting somewhere. I was able to poke around in the DB. Nothing is getting in there. Just system telemetry....

tgalarneau, 2021-04-29:
in MyDelegate add debug print commands for each text_string. That way
can you see what data is coming in. With only 8 cells the 2nd notify for
celldata will be different than for 16. But 1st half doesn't have room for
'77' end of message. Need to see data and length of messages.

class MyDelegate(btle.DefaultDelegate): # notification responses
 def __init__(self):
btle.DefaultDelegate.__init__(self)
def handleNotification(self, cHandle, data):
hex_data = binascii.hexlify(data) # Given raw bytes, get an ASCII string
representing the hex values
text_string = hex_data.decode('utf-8')
if text_string.find('dd04') != -1: # check incoming data for routing to
decoding routines
cellvolts(text_string)
  print(text_string)
elif text_string.find('dd03') != -1:
cellinfo(text_string)

elif text_string.find('77') != -1 and len(text_string) == 38: # x04
cellvolts(text_string)

elif text_string.find('77') != -1 and len(text_string) == 36: # x03
cellinfo(text_string)

tgalarneau, 2021-04-29:
Your data output is still incorrect for cell voltage it is missing 2 cells
after jbdbms and only shows 6 cell values. And totally missing pack info
data. As data integrity checks were built around data length need to check
incoming data.

the-butterfry, 2021-04-29:
can you paste that section if you would. I am just wasting both of our time researching how to do that to the MyDelagate class.

clearly I'm unqualified

class MyDelegate(btle.DefaultDelegate):         # notification responses
        def __init__(self):
                btle.DefaultDelegate.__init__(self)
        def handleNotification(self, cHandle, data):
                hex_data = binascii.hexlify(data)               # Given raw bytes, get an ASCII string representing the hex values
                text_string = hex_data.decode('utf-8')
                #print(message)
                if text_string.find('dd04') != -1:              # check incoming data for routing to decoding routines
                        cellvolts(text_string)
                        #print(message)
                elif text_string.find('dd03') != -1:
                        cellinfo(text_string)
                        #print(message)
                elif text_string.find('77') != -1 and len(text_string) == 38:    # x04
                        cellvolts(text_string)
                        #print(message)
                elif text_string.find('77') != -1 and len(text_string) == 36:    # x03
                        cellinfo(text_string)
                        #print(message)

tgalarneau, 2021-04-29:
The meter flag is not necessary. I use it as a data field to be able to
separate my multiple meters for graphing. In Grafana make sure default db
is influxdb, need also change query fields to your data structure for
select measurement and after where to point data locations

tgalarneau, 2021-04-29:
lass MyDelegate(btle.DefaultDelegate): # notification responses
def *init*(self):
btle.DefaultDelegate.*init*(self)
def handleNotification(self, cHandle, data):
hex_data = binascii.hexlify(data) # Given raw bytes, get an ASCII string
representing the hex values
text_string = hex_data.decode('utf-8')
if text_string.find('dd04') != -1: # check incoming data for routing to
decoding routines
cellvolts(text_string)
print('04-1 ', text_string)
elif text_string.find('dd03') != -1:
cellinfo(text_string)
print(('03-1 ', text_string)
elif text_string.find('77') != -1 and len(text_string) == 38: # x04
cellvolts(text_string)
print(('04 -2 ', text_string)
elif text_string.find('77') != -1 and len(text_string) == 36: # x03
cellinfo(text_string)
print(('03-2 ', text_string)

the-butterfry, 2021-04-29:
yeah, i thought as much with the -m flag. I don need some sort of better live monitoring for the pack, but hadn't gotten to that point in this new install yet. 

I've got all of that connected correctly I think. even sending this over a vpn to the grafana install/rendering location. DB is reading and querying fine in the dashboard. Obviously i'll have to kill the 9-16cells but i was just trying to get something to dump into it at this point.

the-butterfry, 2021-04-29:
*sigh* sorry.. i've got tab/space complaints. I'm not a python coder. 
[snip.txt](https://github.com/tgalarneau/bms/files/6395695/snip.txt)

tgalarneau, 2021-04-29:
Have you managed to get the pack info i.e.  pack
volts,amps,watts,remain,capacity,cycles
data or only cell voltages and 8 good instead of 6 ?

tgalarneau, 2021-04-29:
Gmail won't let me format it correctly. Just put print() directly below
cellvolt(text_string) at same indent. Python is very fussy about
indentation.

tgalarneau, 2021-04-29:
I used tabs not spaces to indent and Python doesn't like mixed tabs and
spaces. A pain in the ass.

the-butterfry, 2021-04-29:
well, got something!

sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
jbdbms,56580,16,3342,3340,3341,3341,3342,3341
04-1  dd0400100d0e0d0c0d0d0d0d0d0e0d0d0d0c0d10
03-1  dd03001b0a71ffe249ab6d600000298a00000000

tgalarneau, 2021-04-29:
That is good. but has no 2nd half.
elif text_string.find('77') != -1 #and len(text_string) == 38: # x04
comment our len check on both x03 and x04 and run again.

the-butterfry, 2021-04-29:
connect 1
sending
03-1  dd03001b0a6eff7f49a16d600000298a00000000
04 -2  000016430308020b4a0b4dfa7277
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
unused,56580,16,3339,3336,3337,3337,3339,3339
04-1  dd0400100d0b0d080d090d090d0b0d0b0d080d0d
04 -2  ff3877
03-1  dd03001b0a6eff7f49a16d600000298a00000000
04 -2  000016430308020b4a0b4dfa7277
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
unused,56580,16,3339,3336,3337,3337,3338,3338
04-1  dd0400100d0b0d080d090d090d0a0d0a0d080d0d
04 -2  ff3a77
03-1  dd03001b0a6eff7349a16d600000298a00000000
04 -2  000016430308020b4a0b4dfa7e77
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
unused,56580,16,3339,3336,3337,3337,3339,3338
04-1  dd0400100d0b0d080d090d090d0b0d0a0d080d0d
04 -2  ff3977
03-1  dd03001b0a6eff8549a16d600000298a00000000
04 -2  000016430308020b4a0b4dfa6c77

tgalarneau, 2021-04-29:
We're getting closer. The cell voltage data is okay. the 2nd half of the
message just contains the checksum and end indicator '77'. My guess is that
your first two corrupt voltages are from the meter field. If you don't use
it you need to remove it from the data message and format completely.For
the cell info data is correct but your 8 cell pack only has 2 temp monitors
and mine 4 for 16 cells. So need to change len to from 36 to 32 at line 137
and line  66.

Test the data and send it to me and if all is well I could update the code
for an 8 cell system.

 m

the-butterfry, 2021-04-29:
yeah... i'm messing with it removing the meter comments and it doesn't seem to clean it up. Pretty sure i've messed up got some extra chars in there still. might be the formatting, but the 56580, 16 is still there after removing meter stuff

connect 1
sending
03-1  dd03001b0a62ff2948f16d600000298a00000000
sending
56580,16,3326,3320,3322,3322,3323,3323ll7,cell8
04-1  dd0400100cfe0cf80cfa0cfa0cfb0cfb0cf80cfe
03-1  dd03001b0a62ff2948f06d600000298a00000000
sending
56580,16,3326,3320,3322,3322,3323,3323ll7,cell8
04-1  dd0400100cfe0cf80cfa0cfa0cfb0cfb0cf80cfe
03-1  dd03001b0a62ff2948f06d600000298a00000000

tgalarneau, 2021-04-29:
To check if not a data problem instead of a format problem add a print() of
cells1 array on line 149. If this shows correct than it's your format.

cells1 = [cell1, cell2, cell3, cell4, cell5, cell6, cell7, cell8]
print(cells1)

the-butterfry, 2021-04-29:
sending
[56580, 16, 3323, 3318, 3320, 3320, 3321, 3321]
56580,16,3323,3318,3320,3320,3321,3321ll7,cell8
04-1  dd0400100cfb0cf60cf80cf80cf90cf90cf60cfb
03-1  dd03001b0a60ff1648ad6d600000298a00000000

tgalarneau, 2021-04-29:
Okay, think I see

if infodata.find('dd03001f') != -1 and len(infodata) == 40:
infodata = (infodata.removeprefix("dd03001f"))
to
if infodata.find('dd0300') != -1 and len(infodata) == 40:
infodata = (infodata.removeprefix("dd0300"))

line 145 celldata = (celldata.removeprefix("dd040020")) change to
celldata = (celldata.removeprefix("dd0400"))

Also need to check length of x03

line 182 elif text_string.find('77') != -1 and len(text_string) == 36: # x03
cellinfo(text_string)
to
elif text_string.find('77') != -1 and # len(text_string) == 36: # x03
cellinfo(text_string)
print(text_string)
print(len(text_string))

tgalarneau, 2021-04-29:
mistake

if infodata.find('dd03001f') != -1 and len(infodata) == 40:
infodata = (infodata.removeprefix("dd03001b"))
to
if infodata.find('dd0300') != -1 and len(infodata) == 40:
infodata = (infodata.removeprefix("dd03001b"))

line 145 celldata = (celldata.removeprefix("dd040010")) change to
celldata = (celldata.removeprefix("dd040010"))

tgalarneau, 2021-04-29:
getting tired

if infodata.find('dd03001f') != -1 and len(infodata) == 40:
infodata = (infodata.removeprefix("dd03001f"))
to
if infodata.find('dd03001b) != -1 and len(infodata) == 40:
infodata = (infodata.removeprefix("dd03001b"))

line 145 celldata = (celldata.removeprefix("dd040020")) change to
celldata = (celldata.removeprefix("dd040010"))

the-butterfry, 2021-04-29:
k, waiting for the internet over 3g cell to come back online. I see what happened there.
Thanks for all the help! this is a cool little script that i'm surprised you're the first to tackle.

the-butterfry, 2021-04-29:
well that's different anyway
sending
[3322, 3316, 3318, 3319, 3320, 3319, 3317, 3323]
3322,3316,3318,3319,3320,3319,3317,3323l7,cell8
04-1  dd0400100cfa0cf40cf60cf70cf80cf70cf50cfb
565.79,0.27,152.76,26,653,18491y,cycles
0,0,0,0,0,0,0,0,c05,c06,c07,c08
03-1  dd03001b0a5fff16483b6d600000298a00000000

the-butterfry, 2021-04-29:
here's my cellvolts 

def cellvolts(data):                    # process cell voltages
        global cells1
        celldata = data
        if celldata.find('dd04') != -1 and len(celldata) == 40:
                celldata = (celldata.removeprefix("dd040010"))
                celldata = (binascii.unhexlify(celldata))
                i = 0
                cell1, cell2, cell3, cell4, cell5, cell6, cell7, cell8 = struct.unpack_from('>HHHHHHHH', celldata, i)
                cells1 = [cell1, cell2, cell3, cell4, cell5, cell6, cell7, cell8]       # needed for max, min, delta calculations
                print(cells1)
                message = ("cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8\r%0i,%0i,%0i,%0i,%0i,%0i,%0i,%0i" % (cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8))
                print(message)

tgalarneau, 2021-04-29:
Everything will likely work once find the correct length it should be 32 as
only missing 4 bytes for 2 temps. But that doesn't seem to work.

Yes this is very simple. What took time was sniffing bluetooth and the very
non standard method JBD used.

tgalarneau, 2021-04-29:
You're missing the \n after the \r
message = ("meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8\r\n%s,%0i,
%0i,%0i,%0i,%0i,%0i,%0i,%0i" %
(meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8))

the-butterfry, 2021-04-29:
woo hoo!
sending
cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
3322,3316,3319,3318,3319,3319,3316,3322
04-1  dd0400100cfa0cf40cf70cf60cf70cf70cf40cfa
volts,amps,watts,remain,capacity,cycles
26.55,-2.34,-62.13,183,280,0
c01,c02,c03,c04,c05,c06,c07,c08
0,0,0,0,0,0,0,0
03-1  dd03001b0a5fff1647826d600000298a00000000

tgalarneau, 2021-04-29:
you're one step away from done. All is needed is the right length for the
2nd half of the cell info process to return
protect,vers,percent,fet,cells,sensors,temp1,temp2

elif infodata.find('77') != -1 and len(infodata) == 36:

and here
elif text_string.find('77') != -1 and len(text_string) == 36:
cellinfo(text_string)

the-butterfry, 2021-04-29:
yeah that didn't do it. 
I did figure out the TIG communication issues though. I've got partial data in Grafana save for that second half with temps.
[snip.txt](https://github.com/tgalarneau/bms/files/6396889/snip.txt)

Current data ourput:
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,3319,3314,3317,3316,3317,3317,3314,3320
meter,volts,amps,watts,remain,capacity,cycles
bms,26.53,-2.28,-60.49,175,280,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0

here's what i've got, tell me what you see.

tgalarneau, 2021-04-29:
It's all correct, only missing 2nd half message of pack info.

Their bluetooth messages are sent in 2 parts for each request. The problem
is that the header with packet information and data length are in the first
message but the length it gives are for the 2 combined messages. The second
half has no header, just the data and end of message '77' and a check sum
that is for the 2 combined messages. Not very useful for sending discrete
message packets.

So we have 2 second half messages without any identifying header info. The
way I addressed this is looking for the '77' and length. The difference
between my 16 cells and your 8 cells is data length of 2nd half messages
and the length info in 1st half header.

To fix your 8 cell system we can just ignore the 2nd half of the cell
voltages as it only contains the check sum and end of message '77' with no
data.

For the temps and other data in pack info 2nd half message requires knowing
the data length for identifying it for processing.

You must query this length with "print(len(text_string))" but remember to
first comment out "# len(text_string) == 36" as below to enable it to show.
Then replace  "len(text_string) == 36" with the correct length. There are 2
instances of this.

*Need to check length and change*

line 182  elif text_string.find('77') != -1 and len(text_string) == 36:  #
x03
            cellinfo(text_string)
 to
          elif text_string.find('77') != -1:             #and
len(text_string) == 36:  # x03
            cellinfo(text_string)
            print(text_string)
            print(len(text_string))

the-butterfry, 2021-04-29:
jbdbms,0,0,0,0,0,0,0,0
0000163d0308020b3a0b3bfb5b77

so.. 26?

tgalarneau, 2021-04-29:
Didn't you get a length return ?

If you count yours it is 14 bytes but when I checked  mine and counted it
was 18 bytes but when print(len(tex_string)) it returned 36 and that worked
for me 18 didn't. I don't know why this.

14 data bytes
protect      ver     %     fet     cells     sensors    temp1   temp2
csum end
0000         16     3d     03     08         02            0b3a     0b3b
 fb5b     77

tgalarneau, 2021-04-29:
For the why it's probably this. "The reason you got weird numbers is
because encapsulated in a string is a bunch of other information due to the
fact that strings are actual objects in python."

the-butterfry, 2021-04-29:
Yeah so 28 I mis-counted previously. 

28 in those two locations produced 
    protect,vers,percent,fet,cells,sensors,temp1,temp2 = struct.unpack_from('>HBBBBBHHHH', infodata, i)
struct.error: unpack_from requires a buffer of at least 15 bytes for unpacking 15 bytes at offset 0 (actual buffer size is 13)

I bumped it to 30 and still 
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,3369,3375,3377,3372,3372,3372,3375,3372
meter,volts,amps,watts,remain,capacity,cycles
bms,26.98,6.88,185.62,180,280,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0

tgalarneau, 2021-04-29:
protect,vers,percent,fet,cells,sensors,temp1,temp2 =
struct.unpack_from('>HBBBBBHHHH', infodata, i)

change to

protect,vers,percent,fet,cells,sensors,temp1,temp2 =
struct.unpack_from('>HBBBBBHH', infodata, i)

tgalarneau, 2021-04-29:
> same result. I wondered if that was a thing. how about the balance2 in
> volts, amps, remain, capacity, cycles, mdate, balance1, balance2 =
> struct.unpack_from('>HhHHHHHH', infodata, i)
> volts=volts/100
> amps = amps/100
> capacity = capacity/100
> remain = remain/100
> watts = volts*amps # adding watts field for dbase
> bal1 = (format(balance1, "b").zfill(16))
> c16 = int(bal1[0:1])
> c15 = int(bal1[1:2]) # using balance1 bits for 16 cells
> c14 = int(bal1[2:3]) # balance2 is for next 17-32 cells - not using
> c13 = int(bal1[3:4])
> c12 = int(bal1[4:5]) # bit shows (0,1) charging on-off
> c11 = int(bal1[5:6])
> c10 = int(bal1[6:7])
> c09 = int(bal1[7:8])
> c08 = int(bal1[8:9])
> c07 = int(bal1[9:10])
> c06 = int(bal1[10:11])
> c05 = int(bal1[11:12])
> c04 = int(bal1[12:13])
> c03 = int(bal1[13:14])
> c02 = int(bal1[14:15])
> c01 = int(bal1[15:16])

Your data output show balance correctly
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0

The problem is with a
elif infodata.find('77') != -1 and len(infodata) == 36:
infodata = (infodata.removesuffix("77"))
infodata = (binascii.unhexlify(infodata))
i = 0
protect,vers,percent,fet,cells,sensors,temp1,temp2,temp3,temp4 = struct.
unpack_from('*>HBBBBBHHHH*', infodata, i)
temp1 = (temp1-2731)/10
temp2 = (temp2-2731)/10 # fet 0011 = 3 both on ; 0010 = 2 disch on ; 0001 =
1 chrg on ; 0000 = 0 both off
temp3 = (temp3-2731)/10
temp4 = (temp4-2731)/10
prt = (format(protect, "b").zfill(16)) # protect trigger (0,1)(off,on)
ovp = int(prt[0:1]) # overvoltage
uvp = int(prt[1:2]) # undervoltage
bov = int(prt[2:3]) # pack overvoltage
buv = int(prt[3:4]) # pack undervoltage
cot = int(prt[4:5]) # current over temp
cut = int(prt[5:6]) # current under temp
dot = int(prt[6:7]) # discharge over temp
dut = int(prt[7:8]) # discharge under temp
coc = int(prt[8:9]) # charge over current
duc = int(prt[9:10]) # discharge under current
sc = int(prt[10:11]) # short circuit
ic = int(prt[11:12]) # ic failure
cnf = int(prt[12:13]) # fet config problem

the-butterfry, 2021-04-29:
oh, duh. I'm just missing a data block is all. sorry. I'm getting it slowly.

the-butterfry, 2021-04-30:
ok had some more time to poke at this. I added a few more printouts in the correct places. So definitely 28.

sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,3328,3323,3324,3325,3326,3326,3325,3329
f9a077
6
meter,volts,amps,watts,remain,capacity,cycles
bms,26.60,-1.91,-50.81,187,280,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0
000016430308020b650b69fb0977
28

I truly have no clue. 

placing prints here doesn't change the output at all.
                message = ("meter,protect,percent,fet,cells,temp1,temp2\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1,temp2))
                print(message)
                print(text_string)
                print(len(text_string))
                sock.send(message.encode('utf8'))       # not sending version number or number of temp sensors

Unles you have some breakthrough idea. I'm willing to give you a SSH to this Pi.

tgalarneau, 2021-04-30:
maybe at first easier to send me your modified code. After that can try ssh.

the-butterfry, 2021-04-30:
This is a no printout version of just straight code.

[jbdbms_8cell.txt](https://github.com/tgalarneau/bms/files/6402810/jbdbms_8cell.txt)

tgalarneau, 2021-04-30:
Looks like minor code corrections

line 64  elif infodata.find('77') != -1 and len(infodata) == 28:
line 132 elif text_string.find('77') != -1 and len(text_string) == 28:

move
line 104 -117  cellmin = min(allcells)
                       cellmax = max(allcells)
                       delta = cellmax-cellmin
                       message =
("meter,cellmin,cellmax,delta\r\n%s,%0i,%0i,%0i" %
(meter,cellmin,cellmax,delta))

to line 103 just after send.socket
     and change min(allcells) to min(cells1)
                         max(allcells) to max(cells1)

tgalarneau, 2021-04-30:
should be lines 114 -117

the-butterfry, 2021-04-30:
k so the 28 on the first '77' line 
Traceback (most recent call last):
  File "/home/pi/bms/jbdbms.py", line 151, in <module>
    result = bms.writeCharacteristic(0x15,b'\xdd\xa5\x04\x00\xff\xfc\x77',False)            # write x04 w/o response cell voltages
  File "/usr/local/opt/python-3.9.4/lib/python3.9/site-packages/bluepy-1.3.0-py3.9.egg/bluepy/btle.py", line 566, in writeCharacteristic
    return self._getResp('wr', timeout)
  File "/usr/local/opt/python-3.9.4/lib/python3.9/site-packages/bluepy-1.3.0-py3.9.egg/bluepy/btle.py", line 436, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "/home/pi/bms/jbdbms.py", line 133, in handleNotification
    cellinfo(text_string)
  File "/home/pi/bms/jbdbms.py", line 68, in cellinfo
    protect,vers,percent,fet,cells,sensors,temp1,temp2 = struct.unpack_from('>HBBBBBHHHH', infodata, i)
struct.error: unpack_from requires a buffer of at least 15 bytes for unpacking 15 bytes at offset 0 (actual buffer size is 13)

changed it to 30 and it runs, but still no data.

[jbdbms_8cell.txt](https://github.com/tgalarneau/bms/files/6402933/jbdbms_8cell.txt)

file after changes

tgalarneau, 2021-04-30:
protect,vers,percent,fet,cells,sensors,temp1,temp2 =
struct.unpack_from('>HBBBBBHHHH', infodata, i)
 change to
protect,vers,percent,fet,cells,sensors,temp1,temp2 = struct.unpack_from('>
*HBBBBBHH*', infodata, i)

Put it back to 28 means 28 was working and enter process but faulted at
above.

the-butterfry, 2021-04-30:
omg, i thought i removed the extra temp HH line!
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,3322,3316,3320,3319,3320,3320,3318,3323
meter,volts,amps,watts,remain,capacity,cycles
bms,26.55,-2.34,-62.13,179,280,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
bms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,protect,percent,fet,cells,temp1,temp2
bms,0,64,3,8,18.2,18.3

the-butterfry, 2021-04-30:
aparently still missing high/low/delta though 
[image]

the-butterfry, 2021-04-30:
Fixed, I think this 8 cell code is done. you might want to do some cleanup or even combine the two and have a -Cells switch.

anyhow.. thank you so much for walking me through this.
[jbdbms_8cell.txt](https://github.com/tgalarneau/bms/files/6406899/jbdbms_8cell.txt)
filling out my dashboard no with Solar data from the charger/inverter unit
[image]

tgalarneau, 2021-04-30:
Yours is looking good. I am taking what we learned and have updated a 8
cell version along with clean handling of sockets, just needs a little more
testing. I am looking into making a version that can handle both too.

Happy Graphing.