tgalarneau/bms #4: BMS Interactivity

the-butterfry, 2021-04-30:
Now that we've got reading data sorted out. I know that the App can interact with the BMS as well. I'm not so sure about changing parameters, but there is one feature that might be useful. 
I've noticed a couple of times that the BT connection hangs and the reader code can't connect. Its behaving like there is already something connected to it. The only way around that seems to be to wait for a timeout/disconnect on the BMS side. 

On the cellphone app there is a button on the login page that allows you to 'Fix' the bluetooth. Looks like it's just pushing a reset command to the module. That would be a useful feature to have, with a little logic that hits reset after XX service restarts. 

that will of course cause hell for the App if someone was actually using it, so maybe not automate it.. but have it available as a -r command appended to the service line. That way every time you start the service and a connect is attempted it will kick over the bluetooth.  Or some other non invasive way of checking for BMS availability.  though that might cause a race condition between a command line launch and a service launch of the code.

tgalarneau, 2021-04-30:
I don't know if I have seen this. But when JBD links to bluetooth then it
will not let you connect with another device until disconnected by first
device. It just hangs on waiting for bluetooth. Be sure you have not left
android app connected. Or if want to use android jbdbms.py must be stopped.

the-butterfry, 2021-04-30:
Yep hence the reset. I was doing some heavy code messing-with, a few times my kludgy attempts left a connection hanging. I had to wait some time for a timeout. Your bms code would throw a couldn't connect error. I had to wait till it would accept connection again. It was at that point when I understood why they had the 'fix' button On the app.

tgalarneau, 2021-04-30:
All you have to do is properly exit the app and to be sure, I turn off
bluetooth on the phone. For the Pi just systemctl stop jbdbms.service.

the-butterfry, 2021-05-01:
no i get that. I'm just relaying my experience. There were several times where the module wouldn't accept a connection when the script or service was stopped. It happened after I attempted to run bad code. I suppose it also could have come from the Pi, too not being able to initiate a connection. *shrug*

tgalarneau, 2021-05-01:
My pi was getting real screwy. Using HTOP could see influxdb was maxing out
cpu. I need to increase collection interval on Grafana. Influx was pulling
the system down serving Grafana.

the-butterfry, 2021-05-01:
Didn't you say you were using a Zero-W? 
This is a pi4 4g booting off of a SSD. Not even a slight performance blip here.

the-butterfry, 2021-05-01:
ahh here we go:

pi@Farm-pi:~/bms $ sudo /usr/local/opt/python-3.9.4/bin/python3.9 jbdbms.py -b A4:C1:38:EC:78:4F -i 7 -m bms
Traceback (most recent call last):
  File "/home/pi/bms/jbdbms.py", line 137, in <module>
    bms = bluepy.btle.Peripheral(args.BLEaddress)   # Connect to bms - 2 tries
  File "/usr/local/opt/python-3.9.4/lib/python3.9/site-packages/bluepy-1.3.0-py3.9.egg/bluepy/btle.py", line 411, in __init__
    self._connect(deviceAddr, addrType, iface, timeout)
  File "/usr/local/opt/python-3.9.4/lib/python3.9/site-packages/bluepy-1.3.0-py3.9.egg/bluepy/btle.py", line 462, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral A4:C1:38:EC:78:4F, addr type: public

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/bms/jbdbms.py", line 139, in <module>
    except BTLEException as ex:
NameError: name 'BTLEException' is not defined
pi@Farm-pi:~/bms $ sudo service jbdbms status
● jbdbms.service - bms Service
   Loaded: loaded (/etc/systemd/system/jbdbms.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2021-04-30 21:53:05 PDT; 56s ago
  Process: 10015 ExecStartPre=/bin/sleep 10 (code=killed, signal=TERM)

Apr 30 21:52:58 Farm-pi systemd[1]: Starting bms Service...
Apr 30 21:53:05 Farm-pi systemd[1]: jbdbms.service: Control process exited, code=killed, status=15/TERM
Apr 30 21:53:05 Farm-pi systemd[1]: jbdbms.service: Succeeded.
Apr 30 21:53:05 Farm-pi systemd[1]: Stopped bms Service.
pi@Farm-pi:~/bms $ sudo /usr/local/opt/python-3.9.4/bin/python3.9 jbdbms.py -b A4:C1:38:EC:78:4F -i 7 -m bms
Traceback (most recent call last):
  File "/home/pi/bms/jbdbms.py", line 137, in <module>
    bms = bluepy.btle.Peripheral(args.BLEaddress)   # Connect to bms - 2 tries
  File "/usr/local/opt/python-3.9.4/lib/python3.9/site-packages/bluepy-1.3.0-py3.9.egg/bluepy/btle.py", line 411, in __init__
    self._connect(deviceAddr, addrType, iface, timeout)
  File "/usr/local/opt/python-3.9.4/lib/python3.9/site-packages/bluepy-1.3.0-py3.9.egg/bluepy/btle.py", line 462, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral A4:C1:38:EC:78:4F, addr type: public

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/bms/jbdbms.py", line 139, in <module>
    except BTLEException as ex:
NameError: name 'BTLEException' is not defined
pi@Farm-pi:~/bms $

tgalarneau, 2021-05-01:
They all show failure to connect bluetooth to bms. JBD bluetooth not too
powerfull. Time for you to use gatttool and look at connection and
device to find out why.

the-butterfry, 2021-05-01:
I know its not signal strength. 
I waited 15 minutes and it came up fine.
k, on the gattool. I've seen it but didn't dive in.

tgalarneau, 2021-05-01:
The bluetooth implementation by JBD comes across as a kludge overlay of
their serial interface. Using gatttool sometimes it just won't connect, no
matter how many times try. But have found once connected it usually
maintains the connection. Waiting usually fixes the problem. On my pi zero
it does seem related to cpu, ram and heat. Works better here in morning
when a little cooler or if dial back collection intervals.

I can see that influxdb reserves 193mb out of the 512 available and the cpu
is up to 100% utilization and this utilization is related to the  amount of
data being pushed to grafana. I 'm planning on tryng victoriametrics to
replace influxdb as it uses 10x less ram and is much faster.

tgalarneau, 2021-05-01:
.Attached is a short python script to test connections. Just replace my
xiaoxiang BLE address with yours.

tgalarneau, 2021-05-01:
Fixed for cleaner exiting code. Really can see it's availability and many
times failure to connect.

tgalarneau, 2021-05-01:
I've updated the git files and included one for 8 cells. Added better
socket control and a few other things. I didn't test the 8 cell code so may
need some tweeks.

the-butterfry, 2021-05-02:
just catching up. Got it.. 

> The bluetooth implementation by JBD comes across as a kludge overlay of their serial interface. Using gatttool sometimes it just won't connect, no matter how many times try. But have found once [...]

Yup, that sounds exactly what I'm talking about. The Zero though. You're using a VW Bug, and I'm using a Ferrari pi with this 4g ram and SSD.
this is running Home Assistant with a bunch of other containers.
[image]

uhhh aparently i grabbed a 2g Pi... Well ok then. I had a bunch laying around. but still. 

Trying the new code now.

the-butterfry, 2021-05-02:
the 16ell code works on mine, but the 8 doesn't. I'm not able to parse the error though... digging into it now side by side. 

16cell BMS code

pi@Farm-pi:~/bms $ sudo /usr/local/opt/python-3.9.4/bin/python3.9 jbdbms.py -b A4:C1:38:EC:78:4F -i 7 -m bms
attempting to connect
connected  A4:C1:38:EC:78:4F
Created socket
sending
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,56580,16,3315,3310,3312,3311,3312,3311
sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,56580,16,3315,3310,3312,3311,3312,3311

8 cell code

pi@Farm-pi:~/bms $ sudo /usr/local/opt/python-3.9.4/bin/python3.9 jbdbms.py -b A4:C1:38:EC:78:4F -i 7 -m bms
Traceback (most recent call last):
  File "/home/pi/bms/jbdbms.py", line 146, in <module>
    class MyDelegate(btle.DefaultDelegate):                 # notification responses
NameError: name 'btle' is not defined

tgalarneau, 2021-05-02:
I just fixed that.

the-butterfry, 2021-05-02:
oh you changed a bunch :)

Yep, i got it too. You just need to add

import atexit

sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,56580,16,3314,3309,3312,3310,3311,3311
meter,cellmin,cellmax,delta
bms,16,56580,56564
meter,volts,amps,watts,remain,capacity,cycles
bms,26.48,-2.15,-56.93,149,280,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
bms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,protect,percent,fet,cells,temp1,temp2
bms,0,53,3,8,14.7,14.8

the-butterfry, 2021-05-02:
wait hang on a sec
[image]

tgalarneau, 2021-05-02:
I think it;s now fixed.

the-butterfry, 2021-05-02:
hah! oops. I even looked at that and didn't notice the extra 0 till i compared them.

sending
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
bms,3313,3308,3311,3310,3310,3310,3307,3313
meter,cellmin,cellmax,delta
bms,3307,3313,6
meter,volts,amps,watts,remain,capacity,cycles
bms,26.48,-2.22,-58.79,148,280,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
bms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
bms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,protect,percent,fet,cells,temp1,temp2
bms,0,53,3,8,14.6,14.7

the-butterfry, 2021-05-02:
Do we need:
            #sock = socket.socket(*self._socket_type,self._socket_data)
or is that legacy now?

tgalarneau, 2021-05-03:
#sock = socket.socket(*self._socket_type,self._socket_data)
 Is the best one to use. I just used the actual values as I thought there
could be a problem. But it is used in meters-socket.py without any problems.

I have installed victoriametrics db to replace influxdb. Where influx was
used 80-90 % cpu victoriametrics uses 7-8% and the ram usage is also way
less. It still uses telegraf and grafana just a different port. I'm
impressed. Great for a pi zero.

the-butterfry, 2021-05-04:
> #sock = socket.socket(*self._socket_type,self._socket_data) Is the best one to use. I just used the actual values as I thought there could be a problem. But it is used in meters-socket.py without any problems. 
>
Got it
>I have installed victoriametrics db to replace influxdb. Where influx was used 80-90 % cpu victoriametrics uses 7-8% and the ram usage is also way less. It still uses telegraf and grafana just a different port. I'm impressed. Great for a pi zero.
>
 I'm using a [monitor for my MPPT](https://github.com/johanmeijer/grott) that only takes influx. I'd rather just run a single distribution point. that kind of overhead though, seems perfect for lightweight things like this. 

I was out all weekend and the script ran flawlessly. 
[image]

the-butterfry, 2021-05-09:
ohh, mqqt! nice, i'll give it a go when i'm back in front of my computer. 
I was working that out on my own to take action/notifications on alarms and battery states.

tgalarneau, 2021-05-09:
Yeah, I added mqtt. I'm pleasantly surprised as it uses less ram and
slightly better cpu usage. It's also fast and great to be able to see the
data flow so easily.

I also streamlined the code and it also now runs faster.

the-butterfry, 2021-05-12:
running the latest iteration of the code. I haven't switched over to MQTT yet, but otherwise the 8s version is still working fine.

rege245, 2021-08-12:
it is possible to monitor multiple BMS at same time ? i have 5 battery packs with 5 JBD BMS

tgalarneau, 2021-08-12:
Should be no problem. Just run 5 instances of program. One for each BMS
address.