tgalarneau/bms #5: Connected but no values

53bh, 2021-05-21:
First thank you very much for your script. I have a Liontron LiFePO4 battery which also has an jbd bms. As the battery has only 4 cells I had to comment out the cell voltages part, because there was an exception: 

File "/home/pi/bms/jbdbms-8-mqtt.py", line 117, in cellvolts1
    cell1, cell2, cell3, cell4, cell5, cell6, cell7, cell8 = struct.unpack_from('>HHHHHHHH', celldata, i)
struct.error: unpack_from requires a buffer of at least 16 bytes.

All other values were working without problems and I could read them using MQTT. I will later try to adapt the script to 4 cells.

But now I have to problem, that there is no response / notification from the bms. If I start the script it shows connect but does not read out values. I already tried waiting and trying later with no success.

Do you what the problem could be and how to fix it? I tried resetting bluetooth without success. 

Regards,
Sebastian

tgalarneau, 2021-05-21:
For the cell volts you just need to remove 4 of the H (HHHH) and
delete cell5,cell6,cell7,cell8 and that should work. As to not receiving
any response all I can think of is to make sure nothing else is connected
to bms. If cell phone has bluetooth connection it will not let another
device connect.

53bh, 2021-05-21:
Thank you for your fast reply, I will test it!

53bh, 2021-05-30:
Works very well, thank you very much!

53bh, 2021-05-31:
Works with very well with the solution "For the cell volts you just need to remove 4 of the H (HHHH) and
delete cell5,cell6,cell7,cell8 and that should work."

closed 2021-05-31 by 53bh