tgalarneau/bms #7: no data in influxdb

simwe, 2021-06-16:
Hi, could you post your influxdb and telegraf config files? 

The 4c bms script reads data correct it seems, it prints out the values to the console at least.

telegraf -debug doesnt give anything more than this:

2021-06-16T15:39:24Z I! Starting Telegraf 1.18.3
2021-06-16T15:39:24Z I! Using config file: /etc/telegraf/telegraf.conf
2021-06-16T15:39:25Z I! Loaded inputs: socket_listener
2021-06-16T15:39:25Z I! Loaded aggregators:
2021-06-16T15:39:25Z I! Loaded processors:
2021-06-16T15:39:25Z I! Loaded outputs: file influxdb
2021-06-16T15:39:25Z I! Tags enabled: host=localhost
2021-06-16T15:39:25Z I! [agent] Config: Interval:10s, Quiet:false, Hostname:"localhost", Flush Interval:10s
2021-06-16T15:39:25Z D! [agent] Initializing plugins
2021-06-16T15:39:25Z D! [agent] Connecting outputs
2021-06-16T15:39:25Z D! [agent] Attempting connection to [outputs.influxdb]
2021-06-16T15:39:25Z D! [agent] Successfully connected to outputs.influxdb
2021-06-16T15:39:25Z D! [agent] Attempting connection to [outputs.file]
2021-06-16T15:39:25Z D! [agent] Successfully connected to outputs.file
2021-06-16T15:39:25Z D! [agent] Starting service inputs
2021-06-16T15:39:25Z I! [inputs.socket_listener] Listening on unixgram:///tmp/telegraf.sock

influxdb.service only creates the database.

root@raspberrypi:/home/pi# journalctl -u influxdb.service -f
-- Logs begin at Wed 2021-06-16 13:06:29 CEST. --
Jun 16 17:33:29 raspberrypi influxd[16585]: ts=2021-06-16T15:33:29.246906Z lvl=info msg="opened HTTP access log" log_id=0UmN6A60000 service=httpd path=stderr
Jun 16 17:33:29 raspberrypi influxd[16585]: ts=2021-06-16T15:33:29.248096Z lvl=info msg="Listening on HTTP" log_id=0UmN6A60000 service=httpd addr=[::]:8086 https=false
Jun 16 17:33:29 raspberrypi influxd[16585]: ts=2021-06-16T15:33:29.248570Z lvl=info msg="Starting retention policy enforcement service" log_id=0UmN6A60000 service=retention check_interval=30m
Jun 16 17:33:29 raspberrypi influxd[16585]: ts=2021-06-16T15:33:29.254453Z lvl=info msg="Listening for signals" log_id=0UmN6A60000
Jun 16 17:33:29 raspberrypi influxd[16585]: ts=2021-06-16T15:33:29.255397Z lvl=info msg="Storing statistics" log_id=0UmN6A60000 service=monitor db_instance=_internal db_rp=monitor interval=10s
Jun 16 17:33:29 raspberrypi influxd[16585]: ts=2021-06-16T15:33:29.269801Z lvl=info msg="Sending usage statistics to usage.influxdata.com" log_id=0UmN6A60000
Jun 16 17:33:59 raspberrypi influxd[16585]: ts=2021-06-16T15:33:59.007488Z lvl=info msg="Executing query" log_id=0UmN6A60000 service=query query="CREATE DATABASE battery"
Jun 16 17:33:59 raspberrypi influxd[16585]: [httpd] 127.0.0.1 - - [16/Jun/2021:17:33:58 +0200] "POST /query HTTP/1.1 {'q': 'CREATE DATABASE "battery"'}" 200 57 "-" "Telegraf/1.18.3 Go/1.16.2" 44e20e9b-ceb8-11eb-8001-b827eb19e23b 38390
Jun 16 17:39:25 raspberrypi influxd[16585]: ts=2021-06-16T15:39:25.118650Z lvl=info msg="Executing query" log_id=0UmN6A60000 service=query query="CREATE DATABASE battery"
Jun 16 17:39:25 raspberrypi influxd[16585]: [httpd] 127.0.0.1 - - [16/Jun/2021:17:39:25 +0200] "POST /query HTTP/1.1 {'q': 'CREATE DATABASE "battery"'}" 200 57 "-" "Telegraf/1.18.3 Go/1.16.2" 0744e89a-ceb9-11eb-8002-b827eb19e23b 17448

tgalarneau, 2021-06-16:
Looks like telegraf is working and listening but influxdb not receiving the
data.

I assume you are using the sockets version. Here is my telegraf
configuration file. As I'm now using victoria-metrics instead of influxd my
port is urls = ["http://127.0.0.1:8428"]. For influx it should be *urls =
["http://127.0.0.1:8086 <http://127.0.0.1:8086>"]* and your password and
username must be input.

The only other thing is to make sure to first create the influxdb and test
it is working using influx command line utility. I don't remember changing
anything in the influxdb configuration file. Make sure to uncomment
#reporter.send_data(message)lines
in script.

For using InfluxDB you need to create a new database and a user with
password.

Open a terminal and on command line type :

   - influx
   - CREATE DATABASE battery
   - influx user create -n "username" -p "password" -o "org-name"
   - exit

simwe, 2021-06-16:
Ah, I never uncommented the reporter.end lines. That figures. I glossed by that part completely, as usual. 
Managed to compile and use victoriametrics as well, although i havnt figured out how to query the victoriametrics database yet. Switched the datasource in grafana but not sure if im seeing old influx data or data from victoriametrics.

Ill probably keep influxdb but outsource it to a new raspberry pi 4, and just have telegraf and the python script on the zero w. Seems to exist more info/examples for a noob like me for influx db. 
 
 Thanks for the awesome code!

tgalarneau, 2021-06-16:
In grafana to use victoria-metrics choose prometheus as the database. The
thing about victoria-metrics is that is mostly invisible, not even
necessary to setup database first.

simwe, 2021-06-17:
Adding victoria metrics in grafana as prometheus doesnt give me any data. Testing it and saving give a "ok" green result in grafana.

Am i supposed to see the write requests in the  victoriametrics service log as i can with prometheus?

root@raspberrypi:/home/pi# journalctl -u victoriametrics.service -f
-- Logs begin at Wed 2021-06-16 23:04:08 CEST. --
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.154Z        info        lib/storage/partition.go:1563        opened part "/data/data/small/2021_06/384_48_20210616230056.542_20210616230207.539_16892EC2309CE553" in 0.007 seconds
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.167Z        info        lib/storage/partition.go:1563        opened part "/data/data/small/2021_06/384_48_20210616225935.498_20210616230046.463_16892EC2309CE54B" in 0.007 seconds
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.179Z        info        lib/storage/partition.go:1563        opened part "/data/data/small/2021_06/8460_141_20210616135620.000_20210616140610.000_1689140A10CE43E9" in 0.008 seconds
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.190Z        info        lib/storage/partition.go:1563        opened part "/data/data/small/2021_06/384_48_20210616230217.679_20210616230328.678_16892EC2309CE55B" in 0.007 seconds
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.251Z        info        app/vmstorage/main.go:101        successfully opened storage "/data" in 0.948 seconds; partsCount: 19; blocksCount: 1518; rowsCount: 62342; sizeBytes: 102578
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.275Z        info        app/vmselect/promql/rollup_result_cache.go:99        loading rollupResult cache from "/data/cache/rollupResult"...
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.307Z        info        app/vmselect/promql/rollup_result_cache.go:125        loaded rollupResult cache from "/data/cache/rollupResult" in 0.029 seconds; entriesCount: 0, sizeBytes: 0
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.310Z        info        app/victoria-metrics/main.go:62        started VictoriaMetrics in 1.008 seconds
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.313Z        info        lib/httpserver/httpserver.go:82        starting http server at http://:8428/
Jun 17 01:03:46 raspberrypi victoriametrics[6673]: 2021-06-16T23:03:46.316Z        info        lib/httpserver/httpserver.go:83        pprof handlers are exposed at http://:8428/debug/pprof/

Telegraf gets info ok

root@raspberrypi:/home/pi# journalctl -u telegraf.service -f
-- Logs begin at Thu 2021-06-17 09:32:47 CEST. --
jun 17 11:47:08 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms fet=3i,cells=4i,temp1=25,temp2=26,protect=0i,percent=85i 1623923225589695483
jun 17 11:47:08 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms cell1=3342i,cell2=3336i,cell3=3332i,cell4=3339i 1623923225612751376
jun 17 11:47:08 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms cellmax=3342i,delta=10i,cellmin=3332i 1623923225618335350
jun 17 11:47:18 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms amps=0,watts=0,remain=51i,capacity=60i,cycles=2i,volts=13 1623923235692869466
jun 17 11:47:18 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms c06=0i,c07=0i,c08=0i,c01=0i,c02=0i,c03=0i,c04=0i,c05=0i 1623923235694344459
jun 17 11:47:18 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms c13=0i,c14=0i,c15=0i,c16=0i,c09=0i,c10=0i,c11=0i,c12=0i 1623923235695899452
jun 17 11:47:18 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms duc=0i,cnf=0i,buv=0i,coc=0i,bov=0i,cot=0i,cut=0i,dot=0i,dut=0i,sc=0i,ovp=0i,uvp=0i,ic=0i 1623923235701545426
jun 17 11:47:18 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms protect=0i,percent=85i,fet=3i,cells=4i,temp1=25,temp2=26 1623923235706007405
jun 17 11:47:18 raspberrypi telegraf[3327]: battpack,host=localhost,meter=bms cell1=3342i,cell2=3337i,cell3=3333i,cell4=3339i 1623923235737991256

tgalarneau, 2021-06-17:
journalctl -u victoriametrics.service -f doesn't give much info for my
install.

Did you change the telegraf configuration to "urls = http://127.0.0.1:8428"
?