tgalarneau/bms #9: Not getting cellinfo2 data

4thbit, 2021-10-05:
Great program! Just got everything configured and running on a Pi3 and it works well!  Had one issue though that I'm not sure about.

I just got my BMS a few months ago.  While using your program I wasn't getting any of the data back that was included in cellinfo2 call so I started looking at the Python program.  It looks like the string you were expecting to use in your delegate had a length of 36 for the second '77' search.  I put a print statement in the for text_string and found that mine is only returning 32 characters.  I changed it and it appears to be working fine now.  Here is the return string that I get on mine. 

000020630310030b880b8b0b8cfb0277

Do you think this is a configuration parameter I set in my BMS somewhere or is the protocol different? It looks like it's all the protect data that's missing.

4thbit, 2021-10-05:
Wait.... I bet it's that temp4 since it doesn't exist on the newer BMS-16S units.  It now only has 3 sensors.  One internal and two external.  Let me try deleting that info from the structure.

4thbit, 2021-10-05:
OK.  I think I see what is being returned.  The 2-Byte protect is empty "all zeroes" and the fourth temp is just an invalid number.  Perhaps you just had the wrong count in there for your string.  Still not sure why no protect data.  Unless this isn't supported.  This is what the breakdown of my data right (unpack specifies the byte size).  

//Data in hex values. 

protect = 0000 
vers = 20 
percent = 63 
fet = 03 
cells = 10
sensors = 03 
temp1 = 0b88
temp2 = 0b8b 
temp3 = 0b8c 
temp4 = fb02 //invalid
endbyte = 77

4thbit, 2021-10-05:
OK.  Now I'm understanding the protocol.  The protect bytes are just showing whether or not that protection is active.  Like when you get a cell overvoltage error.  Did you see this protocol document? It has the data length in the protocol.  Would probably be better to queue the response rather than treat the incoming as two separate strings.  Or maybe set a flag after the first 03 or 04 so that you know the next incoming data is just a continuation. 

https://www.dropbox.com/s/03vfqklw97hziqr/%E9%80%9A%E7%94%A8%E5%8D%8F%E8%AE%AE%20V2%20%28%E6%94%AF%E6%8C%8130%E4%B8%B2%29%28Engrish%29.xlsx?dl=0

tgalarneau, 2021-10-05:
Looks like your bms only has 3 temp variables like the 8 cell versions. So
the temp4 variable needs to be eliminated  from the data structure. Protect
only is a bit on/off switch so always zero unless error and as you suggest
needs to save state data for change.

4thbit, 2021-10-05:
So, is below a bug or is your string length actually 36? Mine is 32 and I had to change it to get any data. 

`elif text_string.find('77') != -1 and len(text_string) == 36:	 # x03
			cellinfo2(data)	`

4thbit, 2021-10-05:
Oh...wait.  That's the checksum.  I see...so you should probably change that line to something like 

`elif text_string.find('77') != -1 and len(text_string) > 31: # x03 cellinfo2(data)`

to handle both versions or in range of 32,36 right?

tgalarneau, 2021-10-05:
If only 3 temp sensors then need change
protect,vers,percent,fet,cells,sensors,temp1,temp2,temp3,temp4,b77 =
struct.unpack_from('>HBBBBBHHHHB', infodata, i)
to
protect,vers,percent,fet,cells,sensors,temp1,temp2,temp3,b77 =
struct.unpack_from('>HBBBBBHHHB', infodata, i)
 and eliminate temp4 references.

Your idea may work but as I only have the 4 temp sensor model hard to test.
Problem with this bms is the separation into 2 data packets for each string
without proper header so had problems to identify and filter bad data
except by data length.

icenov, 2022-10-23:
Thankyou both. I had a similar problem with no cellinfo2 data being returned. I have a 16 string JBD model AP20S006.

rectified using [4thbit's hack](https://github.com/tgalarneau/bms/issues/9#issuecomment-934369740) 
```
elif text_string.find('77') != -1 and len(text_string) > 31: # x03 cellinfo2(data)
```

andrewfraley, 2024-12-16:
I have a new AP20S006 v1.5 with 3 temperature sensors setup as a 16s pack.  I'm betting they changed some things in these newer versions.  I modified handleNotification() as such:

```
    def handleNotification(self, cHandle, data):
        hex_data = binascii.hexlify(data) 		# Given raw bytes, get an ASCII string representing the hex values
        text_string = hex_data.decode('utf-8')
        print("text_string: %s" % text_string)
        print("text_string len: %s" % len(text_string))
        if text_string.find('dd04') != -1:		# check incoming data for routing to decoding routines
            print("Cell Volts 1")
            cellvolts1(data)
        elif text_string.find('dd03') != -1:
            print("Cell Info 1")
            cellinfo1(data)
        elif text_string.find('77') != -1 and len(text_string) == 38:	 # x04
            print("Cell Volts 2")
            cellvolts2(data)
        elif text_string.find('77') != -1 and len(text_string) == 36:	 # x03
            print("Cell Info 2")
            cellinfo2(data)
        else:
            print("No match")
        print("\n\n")
```

And the output:

```
text_string: dd03002614810000151346500000314900000000
text_string len: 40
Cell Info 1
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,52.49,0.00,0.00,53,180,0
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,52.49,0.00,0.00,53,180,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0

text_string: 1000571e0110030b8d0b7d0b7c00000046501513
text_string len: 40
No match

text_string: 0000fb0f77
text_string len: 10
No match

text_string: dd0400200cd20cd20cd00cd30cd10cd40cd40cd1
text_string len: 40
Cell Volts 1
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
jbdbms,3282,3282,3280,3283,3281,3284,3284,3281
meter,cell1,cell2,cell3,cell4,cell5,cell6,cell7,cell8
jbdbms,3282,3282,3280,3283,3281,3284,3284,3281

text_string: 0cd00cd40cc40cd20cd40cd20cd10cd2f20c77
text_string len: 38
Cell Volts 2
meter,cell9,cell10,cell11,cell12,cell13,cell14,cell15,cell16
jbdbms,3280,3284,3268,3282,3284,3282,3281,3282
meter,cell9,cell10,cell11,cell12,cell13,cell14,cell15,cell16
jbdbms,3280,3284,3268,3282,3284,3282,3281,3282
meter,mincell,cellsmin,maxcell,cellsmax,delta
jbdbms,cell11,3268,cell6,3284,16
meter,mincell,cellsmin,maxcell,cellsmax,delta
jbdbms,cell11,3268,cell6,3284,16
```

tgalarneau, 2024-12-17:
I  had a version for 4 temp probes but likely same location in data in
cellinfo2. Just need to change the string length to match new length for 3
sets of temp data along with corrected data structure

andrewfraley, 2024-12-17:
I'm guessing cellinfo2 is in:

```
text_string: 1000571e0110030b8d0b7d0b7c00000046501513
text_string len: 40
No match
```

I'll work on sorting it out.  I didn't like how the data is stored in Influx so I'm refactoring the output to write json to the socket and separating the pack data from the cell data.  This will make it much easier to do a dynamic Grafana dashboard that will work regardless of how many cells you have.  I'll share a fork when I'm done.

andrewfraley, 2024-12-17:
I did get this to work.  Remove all the previously mentioned temp4 stuff, then change the last elif to something like:

```
elif len(text_string) == 40:  # x03
            cellinfo2(data)
```