tgalarneau/bms #11: Not able to connect 4S

Kaythaltes, 2022-01-04:
Hi,
I'm trying to us this programme to pull data from my 4s 280Ah battery with a 100A JBD bms. But I'm not able to connect. 
any advise?

pi@openplotter:~/Downloads/bms $ python3 jbdbms4.py -b A4:C1:38:55:35:3F  -i 10 -m jbdbms
attempting to connect
2nd try connect
Traceback (most recent call last):
  File "jbdbms4.py", line 151, in <module>
    bms = Peripheral(args.BLEaddress,addrType="public")
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 391, in __init__
    self._connect(deviceAddr, addrType, iface)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 439, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral A4:C1:38:55:35:3F, addr type: public

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "jbdbms4.py", line 155, in <module>
    bms = Peripheral(args.BLEaddress,addrType="public")
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 391, in __init__
    self._connect(deviceAddr, addrType, iface)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 439, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral A4:C1:38:55:35:3F, addr type: public
pi@openplotter:~/Downloads/bms $

tgalarneau, 2022-01-04:
Can't say for certain. But from experience know that if bms connected to
another device like a cellphone it will not connect with 2nd device.

Kaythaltes, 2022-01-05:
After rebooting the pi I'm able to connect. 
I get an Error creating socket: [Errno 2] No such file or directory but there is data comming in. Some of them are 0 but the important once for me are showing.

pi@openplotter:~/Downloads/bms $ sudo python3 ./jbdbms-4-socket-2temps.py -b A4:C1:38:55:35:3F -i 15 -m jbdbms
attempting to connect
connected  A4:C1:38:55:35:3F
Error creating socket: [Errno 2] No such file or directory
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.26,0.00,0.00,184,272,6
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,protect,percent,fet,cells,temp1,temp2
jbdbms,0,68,3,4,7.1,7.1
meter,cell1,cell2,cell3,cell4
jbdbms,3316,3318,3319,3315
meter,cellmin,cellmax,delta
jbdbms,3315,3319,4
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.26,0.00,0.00,184,272,6
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,0,0,0,0,0,0,0,0,0,0,0,0,0

tgalarneau, 2022-01-05:
The c01 to c16 fields are flags for cell balancing and are alway 0 unless
actively balancing that cells. The ovp to cnf fields are over protection
and other error flags so should always be 0 unless bms is in error state.

The zeros for amps and watts should only exist if system is not producing
or using power.

The socket error is harder to resolve. Need to check if /tmp/telegraf.sock
file exists and its permissions.

Kaythaltes, 2022-01-07:
Okay so the 0 values are normal, great. So the incomming data are correct.

The socket error is indeed because the file /tmp/telegraf.sock does not exist like you mentioned. Only problem is, being new to telegraf, I can't find how to create this file. Do I have to change something in the telegraf.conf ?

tgalarneau, 2022-01-07:
The  /tmp/telegraf.sock is created  by jbdbms-4-socket-2temps.py and this
script is the one giving the error.

You need to check the permissions on /tmp directory to see why this socket
is not being created.