tgalarneau/bms #12: Step by Step install

K-Mil82, 2022-01-16:


desegura, 2022-05-21:
@tgalarneau it will be nice if someone can help here, specially on debugging the walkthrough 🤔
for example in this setup:

JBDBMS     ------->     RPIZ     ------->     RPI4
   BLE                      BLE->WIFI                  WIFI
  (data)              (tgalarneau/bms)           (client)

where a raspberry zero is reading the jbdbms data over bluetooth an piping the data trough mqtt over wifi to another raspberry 4 where the grafana is installed in home assistant

tgalarneau, 2022-05-24:
With so many OS, installed tools and different setups and degrees of basic skill I can't help in a general way but only with specific problems.

Basic method is test components. Start by running jbdbms in terminal. It will show you if it fails to connect to bluetooth device and by first adding print statements to script it will output data to terminal i.e. (print(ret) at appropiate points) and you will be able to see if all is okay to that point. Meaning no problems with bluetooth or script.

Here's an example and showing error in script :

pi@GridMon:~$ sudo python3 jbdbms-4_cell-1_temp.py -b A4:C1:38:B8:EE:D6 -i 10 -m batt_management
attempting to connect
connected  A4:C1:38:B8:EE:D6
Created socket
meter,volts,amps,watts,remain,capacity,cycles
batt_management,13.85,1.66,22.99,396,400,24
meter,c01,c02,c03,c04,c05,c06,c07,c08
batt_management,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
batt_management,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
batt_management,0,0,0,0,0,0,0,0,0,0,0,0,0
Traceback (most recent call last):
  File "jbdbms-4_cell-1_temp.py", line 187, in <module>
    bms.waitForNotifications(5)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "jbdbms-4_cell-1_temp.py", line 157, in handleNotification
    cellinfo2(data)		
  File "jbdbms-4_cell-1_temp.py", line 123, in cellinfo2
    message = ("meter,protect,percent,fet,cells,temp1\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1))
TypeError: not enough arguments for format string
Closed socket

If all okay at this point, check your mqtt broker to see if it's receiving the data. If there check Telegraf data. You can run Telegraf from the terminal with a debug flag and it will show any problems. In Grafana you will easily see if data is being found and output correctly. I don't use Home Assistant so could offer no help if that is the point of error.

desegura, 2022-06-13:
Thanks for the help @tgalarneau