tgalarneau/bms #13: jbdbms-8-mqtt values are missing

Jetiman, 2022-03-12:
Hi,

I have an 8S BMS.

[https://www.lithiumbatterypcb.com/product/6s-7s-8s-9s-10s-li-ion-or-lifepo4-battery-smart-bms-with-20a-30a-40a-50a-60a-constant-discharge-current-with-rs485-and-uart-communication-function/](url)

But I'm missing the values:
watts
mincell,
cellsmin,
maxcell,
cellsmax,
delta

[1647076696756] INFO (1780 on pi-zero): Starting NodeJBD...
[1647076696790] TRACE (1780 on pi-zero): With arguments...
    _: []
    s: "/dev/ttyUSB0"
    serialport: "/dev/ttyUSB0"
    m: "192.1...:1884"
    mqttbroker: "192...:1884"
    l: "trace"
    loglevel: "trace"
    u: "-"
    mqttuser: "-"
    p: "-"
    mqttpass: "-"
    baudrate: 9600
    b: 9600
    mqtttopic: "NodeJBD"
    t: "NodeJBD"
    pollinginterval: 10
    i: 10
    $0: "node-jbd"
[1647076706840] TRACE (1780 on pi-zero): Getting data from Register 3
[1647076706855] TRACE (1780 on pi-zero): Requesting data from BMS...
[1647076706894] TRACE (1780 on pi-zero): Request sent (HEX): 
    0: 0
    1: 0
    2: 3
    3: 0
    4: 0
    5: 0
    6: 77
[1647076706959] TRACE (1780 on pi-zero): Recieved Data from BMS (HEX): 
    0: 221
    1: 3
    2: 0
    3: 27
    4: 10
    5: 166
    6: 9
    7: 54
    8: 96
    9: 11
    10: 109
    11: 96
    12: 0
    13: 0
    14: 44
    15: 81
    16: 0
    17: 0
    18: 0
    19: 0
    20: 0
    21: 0
    22: 40
    23: 88
    24: 3
    25: 8
    26: 2
    27: 11
    28: 133
    29: 11
    30: 97
    31: 251
    32: 184
    33: 119
[1647076706977] TRACE (1780 on pi-zero): Data from is valid!
[1647076707003] TRACE (1780 on pi-zero): Register 3 Data: 
    packV: "27.26"
    packA: "23.58"
    packBalCap: "245.87"
    packRateCap: "280.00"
    packCycles: 0
    packNumberOfCells: 8
    balanceStatus: [
      {
        "cell0": false
      },
      {
        "cell1": false
      },
      {
        "cell2": false
      },
      {
        "cell3": false
      },
      {
        "cell4": false
      },
      {
        "cell5": false
      },
      {
        "cell6": false
      },
      {
        "cell7": false
      }
    ]
    balanceStatusHigh: [
      {
        "cell0": false
      },
      {
        "cell1": false
      },
      {
        "cell2": false
      },
      {
        "cell3": false
      },
      {
        "cell4": false
      },
      {
        "cell5": false
      },
      {
        "cell6": false
      },
      {
        "cell7": false
      }
    ]
    protectionStatus: {
      "singleCellOvervolt": false,
      "singleCellUndervolt": false,
      "packOvervolt": false,
      "packUndervolt": false,
      "chargeOvertemp": false,
      "chargeUndertemp": false,
      "dischargeOvertemp": false,
      "dischargeUndertemp": false,
      "chargeOvercurrent": false,
      "dischargeOvercurrent": false,
      "shortCircut": false,
      "frontEndDetectionICError": false,
      "softwareLockMOS": false
    }
    bmsSWVersion: 40
    packSOC: 88
    FETStatus: {
      "charging": true,
      "discharging": true
    }
    tempSensorCount: 2
    tempSensorValues: {
      "NTC0": "21.75",
      "NTC1": "18.15"
    }
[1647076707022] TRACE (1780 on pi-zero): Connecting to MQTT broker...
[1647076707031] TRACE (1780 on pi-zero): With MQTT options...
    username: "-"
    password: "-"
    clientId: "nodeJBD_"
[1647076708465] TRACE (1780 on pi-zero): Getting data from Register 4
[1647076708473] TRACE (1780 on pi-zero): Requesting data from BMS...
[1647076708483] TRACE (1780 on pi-zero): Request sent (HEX): 
    0: 0
    1: 0
    2: 4
    3: 0
    4: 0
    5: 0
    6: 77
[1647076708597] TRACE (1780 on pi-zero): Publishing data to MQTT...
[1647076708654] TRACE (1780 on pi-zero): Recieved Data from BMS (HEX): 
    0: 221
    1: 4
    2: 0
    3: 16
    4: 13
    5: 81
    6: 13
    7: 81
    8: 13
    9: 80
    10: 13
    11: 84
    12: 13
    13: 79
    14: 13
    15: 75
    16: 13
    17: 76
    18: 13
    19: 84
    20: 253
    21: 8
    22: 119
[1647076708668] TRACE (1780 on pi-zero): Data from is valid!
[1647076708680] TRACE (1780 on pi-zero): Register 4 Data: 
    cell0mV: 3409
    cell0V: "3.41"
    cell1mV: 3409
    cell1V: "3.41"
    cell2mV: 3408
    cell2V: "3.41"
    cell3mV: 3412
    cell3V: "3.41"
    cell4mV: 3407
    cell4V: "3.41"
    cell5mV: 3403
    cell5V: "3.40"
    cell6mV: 3404
    cell6V: "3.40"
    cell7mV: 3412
    cell7V: "3.41"
[1647076708693] TRACE (1780 on pi-zero): Connecting to MQTT broker...
[1647076708697] TRACE (1780 on pi-zero): With MQTT options...
    username: "-"
    password: "-"
    clientId: "nodeJBD_"
[1647076708793] TRACE (1780 on pi-zero): Publishing data to MQTT...`

tgalarneau, 2022-03-12:
As all these values are derived from the cell voltage data that you are
receiving it's hard to tell what is going on from your printout. Send a
terminal printout of the jbdbms-8-mqtt.py script to see what is happening.

Jetiman, 2022-03-12:
how do i do a terminal printout

tgalarneau, 2022-03-12:
From a terminal type
*jbdbms-8-mqtt.py -b xx:xx:xx:xx:xx -i 10 -m bms-name*

*Then copy and paste output. ( you may need sudo).*

tgalarneau, 2022-03-12:
also will need add print(message) and print(message1) to py script in def
cellvolts1(data)

Jetiman, 2022-03-12:
I have found my mistake. I have now stored an mqtt password.

mqtt = paho.Client("control3")      #create and connect mqtt client
mqtt.username_pw_set("USERNAME", "Password")
mqtt.connect(broker,port)

closed 2022-03-12 by Jetiman

tgalarneau, 2022-03-13:
Glad to hear it. Hope it works out for you.