tgalarneau/bms #14: Telegraf Errors

Vanguard4893, 2022-05-24:
I've had this script running for some time now, and I've come across a problem with writing to the InfluxDB database, with field type conflicts:

```
2022-05-24T20:06:01Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:11Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "cycles" on measurement "battpack" is type string, already exists as type integer dropped=13
2022-05-24T20:06:11Z D! [outputs.influxdb] Wrote batch of 14 metrics in 19.666562ms
2022-05-24T20:06:11Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:21Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): field type conflict
2022-05-24T20:06:21Z D! [outputs.influxdb] Wrote batch of 24 metrics in 13.301769ms
2022-05-24T20:06:21Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:31Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "volts" on measurement "battpack" is type integer, already exists as type string dropped=16
2022-05-24T20:06:31Z D! [outputs.influxdb] Wrote batch of 32 metrics in 23.476845ms
2022-05-24T20:06:31Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:41Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "amps" on measurement "battpack" is type float, already exists as type string dropped=14
2022-05-24T20:06:41Z D! [outputs.influxdb] Wrote batch of 28 metrics in 19.175889ms
2022-05-24T20:06:41Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:51Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "amps" on measurement "battpack" is type float, already exists as type string dropped=14
2022-05-24T20:06:51Z D! [outputs.influxdb] Wrote batch of 28 metrics in 29.30013ms
2022-05-24T20:06:51Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:07:01Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "amps" on measurement "battpack" is type float, already exists as type string dropped=12
2022-05-24T20:07:01Z D! [outputs.influxdb] Wrote batch of 24 metrics in 20.640825ms
2022-05-24T20:07:01Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:11Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "cycles" on measurement "battpack" is type string, already exists as type integer dropped=13
2022-05-24T20:06:11Z D! [outputs.influxdb] Wrote batch of 14 metrics in 19.666562ms
2022-05-24T20:06:11Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:21Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): field type conflict
2022-05-24T20:06:21Z D! [outputs.influxdb] Wrote batch of 24 metrics in 13.301769ms
2022-05-24T20:06:21Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:31Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "volts" on measurement "battpack" is type integer, already exists as type string dropped=16
2022-05-24T20:06:31Z D! [outputs.influxdb] Wrote batch of 32 metrics in 23.476845ms
2022-05-24T20:06:31Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:06:41Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "amps" on measurement "battpack" is type float, already exists as type string dropped=142022-05-24T20:07:11Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "volts" on measurement "battpack" is type integer, already exists as type string dropped=14
2022-05-24T20:07:11Z D! [outputs.influxdb] Wrote batch of 28 metrics in 21.390141ms
2022-05-24T20:07:11Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:07:21Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "volts" on measurement "battpack" is type integer, already exists as type string dropped=14
2022-05-24T20:07:21Z D! [outputs.influxdb] Wrote batch of 28 metrics in 19.355835ms
2022-05-24T20:07:21Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:07:31Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "watts" on measurement "battpack" is type integer, already exists as type string dropped=14
2022-05-24T20:07:31Z D! [outputs.influxdb] Wrote batch of 28 metrics in 18.964483ms
2022-05-24T20:07:31Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:07:41Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "remain" on measurement "battpack" is type integer, already exists as type string dropped=14
2022-05-24T20:07:41Z D! [outputs.influxdb] Wrote batch of 28 metrics in 20.251399ms
2022-05-24T20:07:41Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:07:51Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "amps" on measurement "battpack" is type integer, already exists as type string dropped=14
2022-05-24T20:07:51Z D! [outputs.influxdb] Wrote batch of 28 metrics in 21.775762ms
2022-05-24T20:07:51Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
2022-05-24T20:08:01Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "remain" on measurement "battpack" is type integer, already exists as type string dropped=14
2022-05-24T20:08:01Z D! [outputs.influxdb] Wrote batch of 28 metrics in 50.730893ms
2022-05-24T20:08:01Z D! [outputs.influxdb] Buffer fullness: 0 / 10000 metrics
```

I've tried flushing the entire database; reinstalling InfluxDB from scratch & every time I try starting over, I get these errors, but on different fields every time.

Also, I am using an older version -  the latest Git version of the 4-cell script with a single temp sensor doesn't work correctly, I get the following error when run:

```
pi@GridMon:~$ sudo python3 jbdbms-4_cell-1_temp.py -b A4:C1:38:B8:EE:D6 -i 10 -m batt_management
attempting to connect
connected  A4:C1:38:B8:EE:D6
Created socket
meter,volts,amps,watts,remain,capacity,cycles
batt_management,13.85,1.66,22.99,396,400,24
meter,c01,c02,c03,c04,c05,c06,c07,c08
batt_management,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
batt_management,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
batt_management,0,0,0,0,0,0,0,0,0,0,0,0,0
Traceback (most recent call last):
  File "jbdbms-4_cell-1_temp.py", line 187, in <module>
    bms.waitForNotifications(5)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 416, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "jbdbms-4_cell-1_temp.py", line 157, in handleNotification
    cellinfo2(data)		
  File "jbdbms-4_cell-1_temp.py", line 123, in cellinfo2
    message = ("meter,protect,percent,fet,cells,temp1\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1))
TypeError: not enough arguments for format string
Closed socket
```

tgalarneau, 2022-05-24:
How many temperature probes does your device have ? This is the problem as they keep changing the number of probes.

I see you are using code with 1 probe which is where the problem is and you said that earlier this didn't work.

Vanguard4893, 2022-05-25:
Just the one temp probe in my case. I have been using the same code from the last issue, which has worked fine up until I started getting issues with Telegraf - no other changes made.

Oddly the 4 cell 2 temp probe script does run fine.

Vanguard4893, 2022-05-26:
I've also tried a complete reinstall of the Pi doing the logging, latest version of RaspOS (Bullseye). I still get the same field type conflicts from a fresh start.

Cheers

tgalarneau, 2022-05-26:
Are you saying the script is still crashing on cellinfo2 or elsewhere ?

Vanguard4893, 2022-05-27:
The script modified to work with my board works unchanged and spits out all the correct stats from the BMS, but not all the data ends up in the InfluxDB database, due to the field type conflict errors.

This occurs even if I clear the entire database & start logging data from scratch.

Cheers

tgalarneau, 2022-05-28:
Did you try running telegraf from the terminal with the debug flag. Long ago I remember getting that error but can't remember why and how I fixed it. But if was related to telegraf.

Vanguard4893, 2022-05-28:
Hi,
The errors listed in the first post of this issue are from running Telegraf with the Debug flag, I don't get much else unfortunately.

It's definitely a Telegraf thing, but I can't for the life of me work out what might be going on!

Cheers

tgalarneau, 2022-05-29:
The thing about your error message is that it says 'Failed to write metric" then on the next line states 'Wrote batch of 24 metrics in 20.640825ms'.

It seems like the data is written somewhere. A very schizophrenic error message.

sharpmonk, 2022-08-15:
I also have same issue with infulx spitting out the same errors i have been messing with the script for days now burning my head out.
sometimes if i delete the database while its all running i get different data being stored like cellmin cellmax that are not being stored when i make database first then start it

sharpmonk, 2022-08-15:
the only data that ends up in db is in image below 
[image]

sharpmonk, 2022-08-15:
[image]

sharpmonk, 2022-08-15:
[image]

sharpmonk, 2022-08-15:
[image]

tgalarneau, 2022-08-15:
Looks like your are using jbdbms-16 but only have 4 batteries you should use jbdbms-4. You are getting data from your bms as seen in the terminal  but there is a failure along the way to Grafana. You need to check your socket or mqtt which ever you are using and then see if Telegraf is receiving and sending then if influx db is it set up correctly and receiving data.

sharpmonk, 2022-08-16:
i am receiving data to grafana but only partial data.
the socket-4 still shows all 16 i just checked from fresh download

sharpmonk, 2022-08-16:
solar@solar:~/solarstats/bms $ sudo python3 jbdbms-4-socket-2temps.py  -b A4:C1:38:44:D0:44 -i 5 -m jbdbms
attempting to connect
connected  A4:C1:38:44:D0:44
Created socket
meter,cell1,cell2,cell3,cell4
jbdbms,3342,3341,3343,3343
meter,cellmin,cellmax,delta
jbdbms,3341,3343,2
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.37,0.00,0.00,208,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,cell1,cell2,cell3,cell4
jbdbms,3342,3342,3343,3343
meter,cellmin,cellmax,delta
jbdbms,3342,3343,1
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.36,0.00,0.00,208,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0

sharpmonk, 2022-08-16:
the only data making it to DB is the first section 

meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.37,0.00,0.00,208,280,15

the rest never makes it. its very strange like a buffer issue maybe

tgalarneau, 2022-08-16:
I see you're using the version with 2 temperature monitors and you are not getting the temp info. Try the 1temp version.

sharpmonk, 2022-08-16:
sudo python3 jbdbms-4-socket-1temp.py  -b A4:C1:38:44:D0:44 -i 7 -m jbdbms
attempting to connect
connected  A4:C1:38:44:D0:44
Created socket
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.34,0.00,0.00,207,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,cell1,cell2,cell3,cell4
jbdbms,3336,3335,3336,3336
meter,cellmin,cellmax,delta
jbdbms,3335,3336,1
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.34,0.00,0.00,207,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,cell1,cell2,cell3,cell4
jbdbms,3335,3334,3336,3337
meter,cellmin,cellmax,delta
jbdbms,3334,3337,3
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.34,0.00,0.00,207,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0

same results. i do have 2 temp sensors running.

tgalarneau, 2022-08-16:
You are missing the cellinfo2 data (meter,protect,percent,fet,cells,temp1,temp2) and (meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf). For some reason the function is not being called. This is likely the reason influx is chocking.

sharpmonk, 2022-08-16:
yes i did notice that but it is uncemented it just doesn't seem to be collecting that data. or showing it in the print outs

tgalarneau, 2022-08-16:
Put a print line to show why.

class MyDelegate(DefaultDelegate):		    # notification responses
	def __init__(self):
		DefaultDelegate.__init__(self)
	def handleNotification(self, cHandle, data):
		hex_data = binascii.hexlify(data) 		# Given raw bytes, get an ASCII string representing the hex values
		text_string = hex_data.decode('utf-8')  # check incoming data for routing to decoding routines>>>>>

insert after this 
            print(test-string)

then run to see output.

tgalarneau, 2022-08-16:
should be print(text_string)  

my error

sharpmonk, 2022-08-16:
python3 jbdbms-4-socket-2temps.py  -b A4:C1:38:44:D0:44 -i 10 -m jbdbmsattempting to connect
connected  A4:C1:38:44:D0:44
Created socket
dd03001d05370000512c6d60000f2ae200000000
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
0000204a0304030ba70ba80ba5fab977
dd0400080d0a0d090d0b0d0bff9b77
meter,cell1,cell2,cell3,cell4
jbdbms,3338,3337,3339,3339
meter,cell1,cell2,cell3,cell4
jbdbms,3338,3337,3339,3339
meter,cellmin,cellmax,delta
jbdbms,3337,3339,2
meter,cellmin,cellmax,delta
jbdbms,3337,3339,2
dd03001d05370000512c6d60000f2ae200000000
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
0000204a0304030ba70ba80ba5fab977

tgalarneau, 2022-08-16:
Check hex and it looks like they have now added 3 temp sensors.

sharpmonk, 2022-08-16:
yes the android app shows a main temp then the 2 probes

sharpmonk, 2022-08-16:
so i need to add a temp3

tgalarneau, 2022-08-16:
Try changing   

class MyDelegate(DefaultDelegate):		    # notification responses
	def __init__(self):
		DefaultDelegate.__init__(self)
	def handleNotification(self, cHandle, data):
		hex_data = binascii.hexlify(data) 		# Given raw bytes, get an ASCII string representing the hex values
		text_string = hex_data.decode('utf-8')  # check incoming data for routing to decoding routines
		if text_string.find('dd04') != -1:	                             # x04 (1-8 cells)	
			cellvolts1(data)
		elif text_string.find('dd03') != -1:                             # x03
			cellinfo1(data)
		#elif text_string.find('77') != -1 and len(text_string) == 38:	 # x04 (9-16 cells)
		#	cellvolts2(data)

here change ==28 to ==32

	elif text_string.find('77') != -1 and len(text_string) == 28 or len(text_string) == 36:	 # x03
			cellinfo2(data)	

run to see output if works to enter fuctio will still need another chage

sharpmonk, 2022-08-16:
jbdbms,0,0,0,0,0,0,0,0
0000204a0304030ba30ba60ba4fac177
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,protect,percent,fet,cells,temp1,temp2
jbdbms,0,74,3,4,24.8,25.1
meter,protect,percent,fet,cells,temp1,temp2
jbdbms,0,74,3,4,24.8,25.1

ok that worked now there is just 1 set of hex

tgalarneau, 2022-08-16:
From the hex it appears 3 temps are output just need to change a little code. If it now outputs mangled temp data I will show you the needed changes.

sharpmonk, 2022-08-16:
am all ears haha

tgalarneau, 2022-08-16:
Here need changes

def cellinfo2(data):
    infodata = data  
    i = 0                          # unpack into variables, ignore end of message byte '77'
    protect,vers,percent,fet,cells,sensors,temp1,temp2,b77 = struct.unpack_from('>HBBBBBHHB', infodata, i)

change to
protect,vers,percent,fet,cells,sensors,temp1,temp2,temp3,b77 = struct.unpack_from('>HBBBBBHHHB', infodata, i)

then

   temp1 = (temp1-2731)/10
    temp2 = (temp2-2731)/10	
add
     temp3=(temp3-2731)/10

finally

message = ("meter,protect,percent,fet,cells,temp1,temp2\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1,temp2))

change to

message = ("meter,protect,percent,fet,cells,temp1,temp2,temp3\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1,temp2,temp3))

then run and show output

sharpmonk, 2022-08-16:
dd03001d05370000512a6d60000f2ae200000000
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
0000204a0304030ba20ba60ba2fac577
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,0,0,0,0,0,0,0,0,0,0,0,0,0
meter,protect,percent,fet,cells,temp1,temp2,temp3
jbdbms,0,74,3,4,24.7,25.1,24.7
meter,protect,percent,fet,cells,temp1,temp2,temp3
jbdbms,0,74,3,4,24.7,25.1,24.7
dd0400080d0b0d0b0d0d0d0cff9577
meter,cell1,cell2,cell3,cell4
jbdbms,3339,3339,3341,3340
meter,cell1,cell2,cell3,cell4
jbdbms,3339,3339,3341,3340
meter,cellmin,cellmax,delta
jbdbms,3339,3341,2
meter,cellmin,cellmax,delta
jbdbms,3339,3341,2

sharpmonk, 2022-08-16:
it is showing the 3 temps

tgalarneau, 2022-08-16:
all looks good except you have double output at least on print to terminal.

sharpmonk, 2022-08-16:
only
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.35,0.00,0.00,207,280,15
is making it to the database

tgalarneau, 2022-08-16:
I guess temp sensors are cheap so they keep adding them and because of their bad way of coding bluetooth output makes the script need to be updated.

sharpmonk, 2022-08-16:
2022-08-16T02:01:38Z E! [outputs.influxdb] E! [outputs.influxdb] Failed to write metric (will be dropped: 400 Bad Request): partial write: field type conflict: input field "amps" on measurement "battpack" is type string, already exists as type float dropped=26

is it just the temps that is the issue or is there more to it.

tgalarneau, 2022-08-16:
We only fixed the script output and know it now works. The next 2 programs need to be checked for correctness Telegraf and Influxb.   The flow is BMS - script - socket or mqtt - telegraf - influxdb - grafana. It is easier to check next step with mqtt. However you can run telegraf in debug mode to see is it is okay.

sharpmonk, 2022-08-16:
that output is from telegraf in debug mode. 
i could setup mqtt tomorrow and try that.

sharpmonk, 2022-08-16:
thank you for your help so far. :) much appreciated

sharpmonk, 2022-08-16:
so after much messing today i have the MQTT version modified to 3temps and i am getting the temps through to grafana see image below.

[image]

and after adding some Prints to the script i have this.

```
dd03001d053900005c756d60000f2ae200000000
{'meter': 'bms', 'volts': 13.37, 'amps': 0.0, 'watts': 0.0, 'remain': 236.69, 'capacity': 280.0, 'cycles': 15}
-------
{'meter': 'bms', 'c16': 0, 'c15': 0, 'c14': 0, 'c13': 0, 'c12': 0, 'c11': 0, 'c10': 0, 'c09': 0, 'c08': 0, 'c07': 0, 'c06': 0, 'c05': 0, 'c04': 0, 'c03': 0, 'c02': 0, 'c01': 0}
---------
000020550304030ba40ba20b9ffa6777
{'meter': 'bms', 'ovp': 0, 'uvp': 0, 'bov': 0, 'buv': 0, 'cot': 0, 'cut': 0, 'dot': 0, 'dut': 0, 'coc': 0, 'duc': 0, 'sc': 0, 'ic': 0, 'cnf': 0}
------------
{'meter': 'bms', 'protect': 0, 'percent': 85, 'fet': 3, 'cells': 4, 'temp1': 24.9, 'temp2': 24.7, 'temp3': 24.4}
----------
dd0400080d110d100d100d11ff8277
{'meter': 'bms', 'cell1': 3345, 'cell2': 3344, 'cell3': 3344, 'cell4': 3345}
{'meter': 'bms', 'mincell': 2, 'cellsmin': 3344, 'maxcell': 1, 'cellsmax': 3345, 'delta': 1}

```

so far so good.
However as you can see from the grafana image i am not getting the voltage/ampage data.
i can run the socket and get voltage but not temps or i run mqtt and get other stats but not voltages

all the information is there in the output from the scripts but only half of the data makes it to the database 

im stumped haha 

could you share a telegraf config

tgalarneau, 2022-08-16:
The telegraf config is available on site under configuration. Did you try and edit the voltage and amps data in grafana edit ?

All the data is seems available but sometimes Grafana changes database coding and you need to make sure it is correct for graph.

Last resort you could delete and recreate influxdb data.

sharpmonk, 2022-08-16:
i have droped the DB many times haha but i have just done it again since temp3 fix and the mqtt is working WOOHOO like finally... thank you very much for your support.

[image]

`022-08-16T21:46:55Z D! [outputs.influxdb] Wrote batch of 7 metrics in 31.890811ms
2022-08-16T21:46:55Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:47:05Z D! [outputs.influxdb] Wrote batch of 6 metrics in 17.824924ms
2022-08-16T21:47:05Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:47:15Z D! [outputs.influxdb] Wrote batch of 12 metrics in 7.998587ms
2022-08-16T21:47:15Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:47:25Z D! [outputs.influxdb] Wrote batch of 4 metrics in 14.634637ms
2022-08-16T21:47:25Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:47:35Z D! [outputs.influxdb] Wrote batch of 8 metrics in 8.160788ms
2022-08-16T21:47:35Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:47:45Z D! [outputs.influxdb] Wrote batch of 12 metrics in 11.65333ms
2022-08-16T21:47:45Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:47:55Z D! [outputs.influxdb] Wrote batch of 6 metrics in 26.01486ms
2022-08-16T21:47:55Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:48:05Z D! [outputs.influxdb] Wrote batch of 10 metrics in 6.952417ms
2022-08-16T21:48:05Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:48:15Z D! [outputs.influxdb] Wrote batch of 8 metrics in 8.736465ms
2022-08-16T21:48:15Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
2022-08-16T21:48:25Z D! [outputs.influxdb] Wrote batch of 12 metrics in 6.325426ms
2022-08-16T21:48:25Z D! [outputs.influxdb] Buffer fullness: 0 / 20000 metrics
`

sharpmonk, 2022-08-16:
[jbdbms-4-mqtt-3temps.zip](https://github.com/tgalarneau/bms/files/9354926/jbdbms-4-mqtt-3temps.zip)

marcofesto, 2022-10-14:
> [jbdbms-4-mqtt-3temps.zip](https://github.com/tgalarneau/bms/files/9354926/jbdbms-4-mqtt-3temps.zip)
I had the same issue as you but with your modified version now everything works, Many thanks! (JBD 200A LiFePo4 4S with 3 temp sensors)