tgalarneau/bms #16: Help Running as a Service

gellscom, 2023-01-01:
Hi your work has been a great help and thanks for your efforts. I am a novice at this but have so far suceeded in getting a modified 4 cell  three temp MQTT version running with three JBD BMS's feeding info back into my home assistant, this works fine when launched fron a terminal window using the following command.

cd /home/gerry
python3 BMS_Good.py -b 70:3E:97:08:04:EF -i 10 -m bms-a -q client1 &
python3 BMS_Good.py -b 70:3E:97:07:FA:2E -i 10 -m bms-b -q client2 &
python3 BMS_Good.py -b A4:C1:37:20:8C:CA -i 10 -m bms-c -q client3 &

However try as i might I cannot get it to run from a service. I have limited it to a single bms and tried a number of variations based on your service. I have no idea what i am doing wrong any suggestions would be greatly appreciated. The latest incarnation of my service attached for scrutiny.

I did notine errors refering to the loading of paho in systemctl status enquiry if that helps.

I am sure its a dumb mistake on my part but any help woud be greatly appreciated.

[batt_b .txt](https://github.com/tgalarneau/bms/files/10328683/batt_b.txt)

tgalarneau, 2023-01-02:
I saw this happening before. It was related to loading python modules. I'm no python expert. but found it related to sudo and it worked after I installed paho as superuser alongside normal user. After this it was able to load services. Your service is the same as mine so it should work.

gellscom, 2023-01-02:
Hi thanks ever so much the services are working now.

I have one other issue you may be able to help with,  when modifying stuff to get to three temp sensors and MQTT I madified cellinfo 2 based on some remarks in the issues section of your Github and got the three temps working fine but the protections in message 1 dont seem to work.  When I get to battery over voltage for instance the message does not change from "0" the fet status correctly reports so i think the BMS is ok I must have messed something up.

I have attached my version of you code  if you get a moment can you have a look and see if something obvious jumps out of the page please.

Sorry to be a pain but im almost there.
[BMS_Good.txt](https://github.com/tgalarneau/bms/files/10331859/BMS_Good.txt)

tgalarneau, 2023-01-03:
I have never needed the protection information so never tested it. Not sure but maybe is a temporary flag that would require monitoring and logging over time.