tgalarneau/bms #19: Will this run on Banana Pi BPI-M2 Zero

madmacks59, 2023-03-02:
Based on the constraints on getting RPI Zero 2W boards I'm hoping I can use a Banana Pi BPI-M2 Zero board. Can anyone confirm that this will work?

tgalarneau, 2023-03-03:
I use a Orangepi Zero and it works great. If board runs linux and has bluethooth and wifi it should be okay. In past had some problems with underpowered boards if using multiple devices and mqtt.

madmacks59, 2023-03-03:
I just ordered 4 Orange Pi Zero 2 (1gb) boards based on your comment. With antenna, enclosures, shipping, and tax the total was $135.44 total or $33.89. Higher than I wanted, but I need them now and the Raspberry option is completely out of stock everywhere, or being sold at scalper prices. Thanks for the recommendation!