tgalarneau/bms #20: Can't connect to specific BMS

madmacks59, 2023-03-03:
Im getting a strange result... I have three JBD 100amp BMS', all with the same firmware rev (20). Im running the python code on a 3b+ that is sitting roughly 10 feet from one of my BMS' and maybe 8 feet from the other two. I am able to connect to the two BMS's that are around 8 feet away but not to the one that is 10 feet away. All three should be in close enough procimity to the Pi that Bluetooth range shouldn't be an issue (I will move the mounted 3b+ closer tomorrow to test this). To connect to the BMS' and read data a PIN should not be required. But, on the BMS I can't connect to a PIN was set, and on the other two no PIN has been set. I don't see a way to delete a PIN from the BMS using the iPhone app for the BMS, but I'm wondering if a PIN being defined could  be preventing the python code from connecting.

Two questions. 1) Is the range for Bluetooth on 3b+ PIs very limited? 2) Is there some connection issue that occurs if a BMS has had a PIN set? Normally, you only only need to enter a PIN when your changing the configuration, but as far as I can tell that's the only difference between the three BMS'.

madmacks59, 2023-03-03:
Well, again, never mind...I found the problem. I use XiaoxiangBMS software to communicate with the numerous BMS' I have. That software displays the Bluetooth address for each BMS it can see. It's labeled MAC address in the software. That was the Bluetooth address I was using to try and connect. Turns out that was NOT the actual bluetooth address, even though it worked on two of the three BMS' I was testing with. I ran a bluetooth scanner and it reported a DIFFERENT bluetooth address, which I plugged in, and now I can connect to the BMS I referenced in the first post. I am looking to see why the XiaoxiangBMS software is reporting addresses they way it does. But, I no longer trust the addresses it shows and will use the bluetooth scanned address moving forward. Posting just in case others have a similar issue...

One other oddity. The firmware version is reported as "20" in the XiaoxiangBMS software, but in a different app I use (SmartBMSUtility) it is reported as "32". So one is apparently reporting a Hex value and the other a base 10 value... Wish the chinese apps had decent specs to let people know what is what...

closed 2023-03-03 by madmacks59

tgalarneau, 2023-03-03:
Overkill now has it's own android app for these Xiaoxiang BSM's. It pretty nice and free maybe check it out. The biggest hassle I have is that if the bms is connected one device it will not connect to a second.