tgalarneau/bms #21: Can't see cellinfo2 data on socket or mqtt 4S

dsrawding, 2024-04-02:
When running  jbdbmd-4-socket.py I don't see any of the cellinfo2 data.  Below is the output.  Not sure why it will not create the socket but not worried about that as I really want to send to mqtt.  I've attached a screen shot of the same data making it into mqtt.  

This is a JBD SP22S001 bms that was mfg 2023-12-06.  Not sure if it has anything to with it or not but wanted to let you know.  This is running on a Pi 3B+ with the Victron Venus OS.   Any help is appreciated!

# python3.8 jbdbmd-4-socket.py -b A5:C2:37:08:0A:89 -i 10 -m jbdbms
attempting to connect
connected  A5:C2:37:08:0A:89
Error creating socket: [Errno 2] No such file or directory
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,13.49,0.00,0.00,99,100,2
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,1,0,1,1,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,cell1,cell2,cell3,cell4
jbdbms,3349,3342,3354,3454
meter,cellmin,cellmax,delta
jbdbms,3342,3454,112

[image]

dsrawding, 2024-04-02:
There was no draw on the battery and I had just finished fully charging it.  The data that was being reported is correct.  I am not worried about the socket as I plan to use mqtt which is also working.  

What I don't see is any of the under/over voltage, discharge, current, etc. data.  Also don't see the temp.  It seems that cellinfo1 and cellvolts1 are reporting but nothing from cellinfo2?  Do you have any ideas or things to try to get the additional data?

Thanks for the great code and I'm happy with the data I get.  But if I'm able to get the rest of the data that would be great.

tgalarneau, 2024-04-02:
Does your 4 cell bms have one or two temp probes ? You don't have to use
sockets,  but can use the mqtt version which just uses mqtt without
sockets. I never tested a 4 cell version as didn't have one.

tgalarneau, 2024-04-02:
These bms have 1 to  4 temps and this changes the data structures so need
the right temp version. This would throw off cellinfo2 data.

dsrawding, 2024-04-02:
There is only one temp probe.  I believe I'm using the correct version of the socket program:

def cellinfo2(data):
    infodata = data  
    i = 0                          # unpack into variables, ignore end of message byte '77'
    protect,vers,percent,fet,cells,sensors,temp1,b77 = struct.unpack_from('>HBBBBBHB', infodata, i)
    temp1 = (temp1-2731)/10
    #temp2 = (temp2-2731)/10			# fet 0011 = 3 both on ; 0010 = 2 disch on ; 0001 = 1 chrg on ; 0000 = 0 both off

tgalarneau, 2024-04-02:
You could try the 2 temp version. The temp data is what made this code hard
as they kept changing count and this changed how to read the data for
cellinfo2. So had to create different versions.

tgalarneau, 2024-04-02:
Seems to be failing on :
elif text_string.find('77') != -1 and len(text_string) == 24 or
len(text_string) == 36: # x03
cellinfo2(data)

need check raw data and determining size and structure using wireshark or
another ble reader.

result =
bms.writeCharacteristic(0x15,b'\xdd\xa5\x03\x00\xff\xfd\x77',False) # write
x03 w/o response cell info

This should give you the data to view.

tgalarneau, 2024-04-02:
should be result =
bms.writeCharacteristic(0x15,b'\xdd\xa5\x04\x00\xff\xfc\x77',False) # write
x04 w/o response cell voltages for cell info.

dsrawding, 2024-04-02:
Will try tonight and report back.  Thanks

tgalarneau, 2024-04-02:
This bms uses non standard ble. There are 2 read requests and each then
sends two packets but only once per request. The first has the correct
header while the second has no header but only end of message marker (77).

So must have way to recognize and put together the two halves of message.
The first 0x04 request and 0x11 message is okay.

But the second 0x03 request and 0x11 message can fail on the second half as
I used packet length to direct to decode cellinfo2 routine. Here the last
part contains temps and number varies so need to change length to match
number of temp probes. But don't know how this is now in newer bms devices.

class MyDelegate(DefaultDelegate):    # notification responses
def __init__(self):
DefaultDelegate.__init__(self)
def handleNotification(self, cHandle, data):
hex_data = binascii.hexlify(data)            # Given raw bytes, get an
ASCII string representing the hex values
text_string = hex_data.decode('utf-8')     # check incoming data for
routing to decoding routines

if text_string.find('dd04') != -1:                             # x04 (1-8
cells)
     cellvolts1(data)
elif text_string.find('dd03') != -1:                             # x03
     cellinfo1(data)

* elif text_string.find('77') != -1 and len(text_string) == 24 or
len(text_string) == 36: # x03      cellinfo2(data) *

This is likely where fix is needed.

tgalarneau, 2024-04-02:
Perhaps easiest to try replacing 24 with 26 then if still not working try
28 and finally try 30.

*elif text_string.find('77') != -1 and len(text_string) == 24 or
len(text_string) == 36: # x03      cellinfo2(data)*

dsrawding, 2024-04-03:
Replaced 24 all the way through 32 and nothing changed.  :(

I captured some data with Wireshark.  I don't know how to read it but I've attached the capture.  The data to/from the battery is from A5:C2:37:08:0A:89.  I'll continue to play around with different lengths and see if I can get something to work.  

Thanks for all the assistance.
[bts.log](https://github.com/tgalarneau/bms/files/14845847/bts.log)

tgalarneau, 2024-04-03:
Looked at wireshark data and see they changed the format for the cell info
response. It used to come from two packets but now from three packets. The
data is still there in the same format but need to change the routine that
collects the data. The old routine looked for the end of message market to
identify the correct packet but now it is in the middle packet of the three
sent instead of the last one. The first packet is okay and you now receive
the data and the last packet contains nothing needed but now need to
recognize and capture the 2nd one.

tgalarneau, 2024-04-03:
As I can't test without that version of bms can only give ideas.

Perhaps try replace

elif text_string.find('77') != -1 and len(text_string) == 24 or
len(text_string) == 36: # x03
     cellinfo2(data)

with
 elif text_string.find('D6') != -1
     cellinfo2(data)

This should work as 'D6' is your bms version number should always will be
there. But would need to test. You could also and len after finding string
that length.

dsrawding, 2024-04-04:
Got more data before it died!

attempting to connect
connected  A5:C2:37:08:0A:89
Error creating socket: [Errno 111] Connection refused
meter,volts,amps,watts,remain,capacity,cycles
jbdbms,12.78,0.00,0.00,15,100,3
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,1,1,0,1,1,1,0,1,0,0,0,0,0
Traceback (most recent call last):
  File "jbdbmd-4-socket.py", line 190, in <module>
    bms.waitForNotifications(5)
  File "/usr/lib/python3.8/site-packages/bluepy-1.3.0-py3.8.egg/bluepy/btle.py", line 592, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/lib/python3.8/site-packages/bluepy-1.3.0-py3.8.egg/bluepy/btle.py", line 436, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "jbdbmd-4-socket.py", line 159, in handleNotification
    cellinfo2(data)		
  File "jbdbmd-4-socket.py", line 125, in cellinfo2
    message = ("meter,protect,percent,fet,cells,temp1\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1))
TypeError: not enough arguments for format string
Error closing socket: 'StatsReporter' object has no attribute '_sock'

tgalarneau, 2024-04-04:
Other than socket problem.

Seems error in struct unpack

def cellinfo2(data):
    infodata = data
    i = 0                          # unpack into variables, ignore end of
message byte '77'
    protect,vers,percent,fet,cells,sensors,temp1,b77 =
struct.unpack_from('>HBBBBBHB', infodata, i)

CHANGE TO

*protect,vers,percent,fet,cells,sensors,temp1 =
struct.unpack_from('>HBBBBBH', infodata, i)*

dsrawding, 2024-04-04:
Made it one more line.

meter,volts,amps,watts,remain,capacity,cycles
jbdbms,12.73,0.00,0.00,15,100,3
meter,c01,c02,c03,c04,c05,c06,c07,c08
jbdbms,0,0,0,0,0,0,0,0
meter,c09,c10,c11,c12,c13,c14,c15,c16
jbdbms,0,0,0,0,0,0,0,0
meter,ovp,uvp,bov,buv,cot,cut,dot,dut,coc,duc,sc,ic,cnf
jbdbms,1,1,0,1,1,1,0,1,0,0,0,0,0
Traceback (most recent call last):
  File "jbdbmd-4-socket.py", line 191, in <module>
    bms.waitForNotifications(5)
  File "/usr/lib/python3.8/site-packages/bluepy-1.3.0-py3.8.egg/bluepy/btle.py", line 592, in waitForNotifications
    resp = self._getResp(['ntfy','ind'], timeout)
  File "/usr/lib/python3.8/site-packages/bluepy-1.3.0-py3.8.egg/bluepy/btle.py", line 436, in _getResp
    self.delegate.handleNotification(hnd, data)
  File "jbdbmd-4-socket.py", line 160, in handleNotification
    cellinfo2(data)		
  File "jbdbmd-4-socket.py", line 126, in cellinfo2
    message = ("meter,protect,percent,fet,cells,temp1\r\n%s,%0000i,%00i,%00i,%0i,%0.1f,%0.1f" % (meter,protect,percent,fet,cells,temp1))
TypeError: not enough arguments for format string
Error closing socket: 'StatsReporter' object has no attribute '_sock'

tgalarneau, 2024-04-04:
Maybe try the mqtt version for this so it doesn't keep crashing over the
missing socket. Problem seems relao get more info need a print statement
for rcvd string.

with
 elif text_string.find('D6') != -1
     cellinfo2(data)
     print(text_string)

dsrawding, 2024-04-04:
Below is the data reporting in mqtt.  Seems to be missing the temp but everything else looks good.  This is fine with me.

{
  "meter": "jbdbms",
  "mincell": 2,
  "cellsmin": 2950,
  "maxcell": 4,
  "cellsmax": 3150,
  "delta": 200
}

{
  "meter": "bms",
  "volts": 12.29,
  "amps": 0,
  "watts": 0,
  "remain": 15.41,
  "capacity": 100,
  "cycles": 3
}

{
  "meter": "bms",
  "cell1": 3056,
  "cell2": 2950,
  "cell3": 3139,
  "cell4": 3150
}

{
  "meter": "bms",
  "c16": 0,
  "c15": 0,
  "c14": 0,
  "c13": 0,
  "c12": 0,
  "c11": 0,
  "c10": 0,
  "c09": 0,
  "c08": 0,
  "c07": 0,
  "c06": 0,
  "c05": 0,
  "c04": 0,
  "c03": 0,
  "c02": 0,
  "c01": 0
}

tgalarneau, 2024-04-04:
Looks good. I never much used the temp info for anything, nor is the fault
info really useful and is available on the app anyway if it's ever
required. You have all the daily useful data for monitoring your
batteries.  Your delta for your cells looks like need to balance them.

dsrawding, 2024-04-04:
Yes, the battery was 100% discharged and I added a small charge to test your code.  I'm charging it now and it will balance out.  

Thanks for all your help and great software!

closed 2024-04-04 by dsrawding